gsilano/CrazyS #4: Crazyflie Roll/Pitch/Yaw controller

arya-k, 2018-09-15:
Hello! Thank you for putting in all the hard work to build a Crazyflie simulator with Gazebo - without it my research project would be MUCH harder.

I need to use a roll/pitch/yaw/thrust controller with the Crazyflie, and I see that you've emulated the CrazyFlie's builtin controller for your position_controller, but I'm not sure how to use that.

In other words: How could I use the `crazyflie_onboard_controller` so that I can accept roll/pitch/yaw/thrust messages (like those sent with `crazyflie_ros`) and have the model behave just like the actual firmware would make it?

gsilano, 2018-09-17:
Thank you for finding CrazyS interesting for you. As reported in the README.md file, if you are using the simulator in the search for the publication/project, please cite the reference paper. We are writing a further paper to explain in more detail how the code works (it will help developers and researchers to use CrazyS for research) and how to use it to simulate the Crazyflie 2.0

What do you mean exactly with "roll/pitch/yaw/thrust controller"? Did you emulate the joypad?

You should use the _SetControlSignals_, _SetDroneState_ and _RateController_ functions (described in the "_crazyflie_onboard_controller.cpp_" file), as shown in line 185 of the "_position_controller.cpp_" file and in Figure 7 of the reference paper.

arya-k, 2018-09-18:
Perfect! Thank you so much. I'll make sure to cite the reference paper in my research paper.

closed 2018-09-18 by arya-k