gsilano/CrazyS #6: Unknown substitution command [eval enable_state_estimator == false]

PercyJaiswal, 2018-09-27:
Hello team,

I followed all instructions step by step to install CrazyS and get started with Crazyflie development. But whenever I am trying to launch "crazyflie2_hovering_example.launch enable_state_estimator:=true', I am getting below error

"Unknown substitution command [eval enable_state_estimator == false]. Valid commands are ['find', 'env', 'optenv', 'anon', 'arg']
The traceback for the exception was written to the log file"

Can you please help me out here. I am using ROS Indigo. Even if I try to launch crazyflie2_hovering_example.launch without additional argument, I still get same error. 

When I am trying to launch any other file from same folder, its opens up gazebo but gives an error saying "No namespace found"

Appreciate your help
Regards

gsilano, 2018-09-27:
Hi @PercyJaiswal,

are you sure you have set the ROS environment correctly? The simulator should work without any problem.

I mean, maybe you installed ROS in an already functioning environment. Please, follow the ROS instructions (see the README.md file) for your specific version of the operating system (Indigo Igloo, if I understand correctly). I'm thinking this could be the problem because you wrote: in the bash, I see "No namespace found".

 ```
echo`` "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
source ~/.bashrc
 ```

PercyJaiswal, 2018-09-27:
Sir, I have followed exact instructions listed in README file, including "Installation Instructions - Ubuntu 14.04 with ROS Indigo". I have tried installing CrazyS in existing workspace as well as tried creating a new one and then installing. As mentioned in README, i do catkin init and catkin build, followed by two source bash commands mentioned by you.

As I am following all steps correctly, I feel my installation is correct.
One thing I can point out is, when I following response. As you can see, in summary it says 2 packages were skipped or are blacklisted. As this workspace only has CrazyS, all packages are of either crazyS or mav_comm. Apart from this everything goes smooth....and I mentioned before, I am able to open few other files like fixed_wing. The error which I am getting for Crazyflie2 hovering example file, I dont get that with any other launch file. 

Profile:                     default
Extending:             [env] /opt/ros/indigo
Workspace:                   /home/percy/new_catkin_ws
--------------------------------------------------------------
Source Space:       [exists] /home/percy/new_catkin_ws/src
Log Space:         [missing] /home/percy/new_catkin_ws/logs
Build Space:        [exists] /home/percy/new_catkin_ws/build
Devel Space:        [exists] /home/percy/new_catkin_ws/devel
Install Space:      [unused] /home/percy/new_catkin_ws/install
DESTDIR:            [unused] None
--------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
--------------------------------------------------------------
[build] Found '13' packages in 0.0 seconds.                                    
[build] Updating package table.                                                
Starting  >>> catkin_tools_prebuild                                            
Finished  <<< catkin_tools_prebuild                [ 1.0 seconds ]             
Starting  >>> mav_msgs                                                         
Starting  >>> rotors_comm                                                      
Starting  >>> rotors_description                                               
Starting  >>> rotors_evaluation                                                
Finished  <<< rotors_description                   [ 2.0 seconds ]             
Starting  >>> rqt_rotors                                                       
Finished  <<< rotors_evaluation                    [ 2.3 seconds ]             
Finished  <<< rqt_rotors                           [ 2.1 seconds ]             
Finished  <<< rotors_comm                          [ 4.3 seconds ]             
Finished  <<< mav_msgs                             [ 4.7 seconds ]             
Starting  >>> mav_planning_msgs                                                
Starting  >>> rotors_control                                                   
Starting  >>> rotors_hil_interface                                             
Starting  >>> rotors_joy_interface                                             
Finished  <<< rotors_hil_interface                 [ 2.0 seconds ]             
Finished  <<< mav_planning_msgs                    [ 7.3 seconds ]             
Finished  <<< rotors_joy_interface                 [ 7.9 seconds ]             
Finished  <<< rotors_control                       [ 27.9 seconds ]            
Starting  >>> rotors_gazebo_plugins                                            
Finished  <<< rotors_gazebo_plugins                [ 1 minute and 3.0 seconds ]
Starting  >>> rotors_gazebo                                                    
Finished  <<< rotors_gazebo                        [ 9.4 seconds ]             
[build] Summary: All 12 packages succeeded!                                    
[build]   Ignored:   2 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                                                     
[build] Runtime: 1 minute and 46.3 seconds total.                              
[build] Note: Workspace packages have changed, please re-source setup files to use them.

Thanks for helping me out.

PercyJaiswal, 2018-09-28:
Hello sir,
I was looking more into error along with having a look at crazyflie2_hovering_example.launch file. I believe this error has something to do with contents of launch file (and not the installation which we are suspecting). The error message is saying that "Unknown substitution command [eval enable_state_estimator == false]. Valid commands are ['find', 'env', 'optenv', 'anon', 'arg']"

This "eval enable_state_estimator == false" command is also present in crazyflie2_hovering_example.launch launch file at line# 34. Will this act as some kind of clue? I am not that well versed with launch files in ROS, but I will keep trying to look into it more.

Appreciate any help.
Regards

gsilano, 2018-09-30:
I found the issue. As described in the link below, the "eval command" in the launch was released with the Kinetic Kame version of ROS. So, you have to fix the issue by changing something in the XML file or changing the version of ROS you are using.

http://wiki.ros.org/roslaunch/XML

I will try to fix the bug in the next few days if you do not do it before. I will let you informed through this issue.

closed 2018-09-30 by gsilano