gsilano/CrazyS #11: improve simulation real time factor

mahdinobar, 2019-03-14:
I launch _crazyflie2_hovering_example_ from _rotors_gazebo_ package on Ubuntu 16.04. The real time factor of simulation in Gazebo is 0.52 while all my CPU cores are not saturated as shown bellow. I was wondering how could I make simulation to use full processing capacity of my computer in order to improve (increase) real time factor? 
[image]

gsilano, 2019-03-14:
Hi @mahdinobar , it's a very good question. It's hard to say "you should do this", to improve the real-time factor. As you can see [here](https://github.com/ethz-asl/rotors_simulator/issues/423), or [here](https://discourse.ros.org/t/real-time-factor-going-really-low/1875), it is an open question and there isn't a single solution. Maybe someone fixed it, but I didn't find it until now. 
However, a good way to improve the real-time factor is to reduce sensors and real-time plots involved during the simulation. As for the plots, it should be fairly simple: just remove the ones included in the launch file. Instead, for the sensors, you should modify the xacro file by going in detail in the RotorS architecture (the ROS package employed as the basis for CrazyS). More details on how CrazyS was made are presented in the [references publications](https://github.com/gsilano/CrazyS/wiki/Publications). 

If you want to know more, please do not hesitate to write again.

mahdinobar, 2019-04-01:
The real time factor issue in my computer is solved by doing so:

At line 26 of file _basic_crazyflie.world_ in the path  _~/rotors_gazebo/worldsf_ by decreasing number of iterations for ode solver from 1000 to 200, the real time factor increases from 0.52 to 1.

gsilano, 2019-04-01:
Yes, it is a good way to increase the real-time factor but, as described [here](http://sdformat.org/spec?elem=physics), when the number of iterations is decreased, the accuracy also decreases. I suggest you take a look at the many questions available on ROS Wiki and also on StackOverflow.

closed 2019-04-16 by gsilano