gsilano/CrazyS #13: Crazyflie physical model in simulation

juanmed, 2019-05-30:
Hello,

Thanks for sharing CrazyS!  In the README.md you mention "...expands RotorS capabilities by considering the Crazyflie 2.0 physical model...". I was wondering if by this you  mean that  you changed the dynamic equations - equations (1) - (9) for in [1]- modelled in RotorS for drone flight to fit better the motion of Crazyfly?  or you mean that you identified the parameters of the drone and used those parameters but without modifying the dynamic equations RotorS is simulating?

I am interested in changing the dynamic equations RotorS is simulating in order to consider other physical quantities, and wanted to use your repository for guidance. I checked the files but did not find anything similar. If you did changed the dynamic equations, would you mind guiding me on which files to refer to in your repo?

Thanks!

Juan

[1] Koubaa, A. (2017). Robot Operating System (ROS) - The Complete Reference-Vol1 (Vol. 707). https://doi.org/10.1007/978-3-319-54927-9 Part VII : RotorS A Modular Gazebo MAV Simulator Framework

gsilano, 2019-05-30:
Hi @juanmed, thanks for opening an issue on CrazyS. I happy to hear from you that you found it useful for your research purposes.

As for your question, I did not modify the RotorS model, but I adapted it to Crazyflie. I mean, I modified (to be precise, creating a new file to model the nano-quadcopter) the YAML file adapting it to the drone.

If you have other questions, please do not hesitate to write here or to open a new issue.

juanmed, 2019-05-30:
@gsilano  Thanks for your clear and timely answer.

closed 2019-05-30 by juanmed