gsilano/CrazyS #15: Waypoint navigation of crazyflie2

ijibkv, 2019-10-27:
I am trying waypoint navigation of Crazyflie2 using the following code:

<launch>
  <arg name="mav_name" default="crazyflie2"/>
  <arg name="world_name" default="basic"/>
  <arg name="enable_logging" default="false" />
  <arg name="enable_ground_truth" default="true" />
  <arg name="enable_state_estimator" default="false" />
  <arg name="log_file" default="$(arg mav_name)" />
  <arg name="paused" value="true"/>
  <arg name="debug" default="false"/>
  <arg name="gui" default="true"/>
  
  <arg name="verbose" default="false"/>
  
  
  <env name="GAZEBO_MODEL_PATH" value="${GAZEBO_MODEL_PATH}:$(find rotors_gazebo)/models"/>
  <env name="GAZEBO_RESOURCE_PATH" value="${GAZEBO_RESOURCE_PATH}:$(find rotors_gazebo)/models"/>
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" value="$(find rotors_gazebo)/worlds/$(arg world_name)_crazyflie.world" />
    <arg name="debug" value="$(arg debug)" />
    <arg name="paused" value="$(arg paused)"/>
    <arg name="gui" value="$(arg gui)" />
    <arg name="verbose" value="$(arg verbose)"/>
  </include>

  <group ns="$(arg mav_name)">
  
    <include file="$(find rotors_gazebo)/launch/spawn_mav_crazyflie.launch">
      <arg name="mav_name" value="$(arg mav_name)" />
      <arg name="model" value="$(find rotors_description)/urdf/mav_generic_odometry_sensor.gazebo" />
      <arg name="enable_logging" value="$(arg enable_logging)" />
      <arg name="enable_ground_truth" value="$(arg enable_ground_truth)" />
      <arg name="enable_state_estimator" value="$(arg enable_state_estimator)" />
      <arg name="log_file" value="$(arg log_file)"/>
    </include>
   
   <node name="position_controller_node" pkg="rotors_control" type="position_controller_node" output="screen">
      <param name="enable_state_estimator" value="$(arg enable_state_estimator)" />
      <rosparam unless="$(arg enable_state_estimator)" command="load" file="$(find rotors_gazebo)/resource/controller_$(arg mav_name).yaml" />
      <rosparam if="$(arg enable_state_estimator)" command="load" file="$(find rotors_gazebo)/resource/controller_$(arg mav_name)_with_stateEstimator.yaml" />
      <rosparam command="load" file="$(find rotors_gazebo)/resource/$(arg mav_name).yaml" />
   </node>
   <node name="hovering_example" pkg="rotors_gazebo" type="hovering_example" output="screen" />
   <node name="waypoint_publisher_file" pkg="rotors_gazebo" type="waypoint_publisher_file" output="screen" args="$(find rotors_gazebo)/resource/example_waypoints.txt"/>
   <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
   <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher" />
   <node name="quaternion_to_rpy" pkg="rotors_gazebo" type="quaternion_to_rpy" output="screen" >
       <remap from="odometry" to="odometry_sensor1/odometry" />
   </node>
   
   <node name="position_plot" pkg="rqt_plot" type="rqt_plot" args="--clear-config /$(arg mav_name)/odometry_sensor1/odometry/pose/pose/position" output="screen" />
   <node name="orientation_plot" pkg="rqt_plot" type="rqt_plot" args="--clear-config /$(arg mav_name)/orientation_rpy" output="screen" />

  </group>
</launch>

the hovering part is working properly but it is not going two the next points and taking too much time(settling time). pls if someone could help me

ijibkv, 2019-10-27:
> I am trying waypoint navigation of Crazyflie2 using the following code:
> 
> the hovering part is working properly but it is not going two the next points and taking too much time(settling time). pls if someone could help me

ijibkv, 2019-10-27:
I am trying waypoint navigation of Crazyflie2 using the following code:

<launch>
  <arg name="mav_name" default="crazyflie2"/>
  <arg name="world_name" default="basic"/>
  <arg name="enable_logging" default="false" />
  <arg name="enable_ground_truth" default="true" />
  <arg name="enable_state_estimator" default="false" />
  <arg name="log_file" default="$(arg mav_name)" />
  <arg name="paused" value="true"/>
  <arg name="debug" default="false"/>
  <arg name="gui" default="true"/>
  
  <arg name="verbose" default="false"/>
  
  
  <env name="GAZEBO_MODEL_PATH" value="${GAZEBO_MODEL_PATH}:$(find rotors_gazebo)/models"/>
  <env name="GAZEBO_RESOURCE_PATH" value="${GAZEBO_RESOURCE_PATH}:$(find rotors_gazebo)/models"/>
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" value="$(find rotors_gazebo)/worlds/$(arg world_name)_crazyflie.world" />
    <arg name="debug" value="$(arg debug)" />
    <arg name="paused" value="$(arg paused)"/>
    <arg name="gui" value="$(arg gui)" />
    <arg name="verbose" value="$(arg verbose)"/>
  </include>

  <group ns="$(arg mav_name)">
  
    <include file="$(find rotors_gazebo)/launch/spawn_mav_crazyflie.launch">
      <arg name="mav_name" value="$(arg mav_name)" />
      <arg name="model" value="$(find rotors_description)/urdf/mav_generic_odometry_sensor.gazebo" />
      <arg name="enable_logging" value="$(arg enable_logging)" />
      <arg name="enable_ground_truth" value="$(arg enable_ground_truth)" />
      <arg name="enable_state_estimator" value="$(arg enable_state_estimator)" />
      <arg name="log_file" value="$(arg log_file)"/>
    </include>
   
   <node name="position_controller_node" pkg="rotors_control" type="position_controller_node" output="screen">
      <param name="enable_state_estimator" value="$(arg enable_state_estimator)" />
      <rosparam unless="$(arg enable_state_estimator)" command="load" file="$(find rotors_gazebo)/resource/controller_$(arg mav_name).yaml" />
      <rosparam if="$(arg enable_state_estimator)" command="load" file="$(find rotors_gazebo)/resource/controller_$(arg mav_name)_with_stateEstimator.yaml" />
      <rosparam command="load" file="$(find rotors_gazebo)/resource/$(arg mav_name).yaml" />
   </node>
   <node name="hovering_example" pkg="rotors_gazebo" type="hovering_example" output="screen" />
   <node name="waypoint_publisher_file" pkg="rotors_gazebo" type="waypoint_publisher_file" output="screen" args="$(find rotors_gazebo)/resource/example_waypoints.txt"/>
   <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
   <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher" />
   <node name="quaternion_to_rpy" pkg="rotors_gazebo" type="quaternion_to_rpy" output="screen" >
       <remap from="odometry" to="odometry_sensor1/odometry" />
   </node>
   
   <node name="position_plot" pkg="rqt_plot" type="rqt_plot" args="--clear-config /$(arg mav_name)/odometry_sensor1/odometry/pose/pose/position" output="screen" />
   <node name="orientation_plot" pkg="rqt_plot" type="rqt_plot" args="--clear-config /$(arg mav_name)/orientation_rpy" output="screen" />

  </group>
</launch>

the hovering part is working properly but it is not going two the next points and taking too much time(settling time). pls if someone could help me

ijibkv, 2019-10-27:
less="$(arg enable_state_estimator)" command="load" file="$(find rotors_gazebo)/resource/controller_$(arg mav_name).yaml" />
      <rosparam if="$(arg enable_state_estimator)" command="load" file="$(find rotors_gazebo)/resource/controller_$(arg mav_name)_with_stateEstimator.yaml" />
      <rosparam command="load" file="$(find rotors_gazebo)/resource/$(arg mav_name).yaml" />
   </node

ijibkv, 2019-10-27:
<node name="hovering_example" pkg="rotors_gazebo" type="hovering_example" output="screen" />
   <node name="waypoint_publisher_file" pkg="rotors_gazebo" type="waypoint_publisher_file" output="screen" args="$(find rotors_gazebo)/resource/example_waypoints.txt"/>

closed 2019-10-27 by ijibkv

gsilano, 2019-10-27:
Hi! As explained in the [reference papers](https://github.com/gsilano/CrazyS/wiki/Publications), CrazyS is a ROS package that aims to simulate the behavior of the Crazyflie 2.0 nano-quadrotor. The hovering (and not trajectory) controller is one of the controller algorithms that can be used to make it flies. If you want to know something more about how the ROS package works, please read the papers I mentioned and do not hesitate to open an issue in case of problems.