gsilano/CrazyS #16: Update Readme

AKR2470, 2019-10-27:
Please update the readme with instructions for Ubuntu 18 and ROS Melodic.

AKR2470, 2019-10-27:
Checkout the logs after catkin build: [https://gist.github.com/AKR2470/012c17bd2d392df47ae063d837f04ace](url)

gsilano, 2019-10-28:
Hi @AKR2470! First of all, thanks for using CrazyS. Then, as far as your request is concerned, as you can read from the [reference papers](https://github.com/gsilano/CrazyS/wiki/Publications), CrazyS uses as a basis [RotorS](https://github.com/ethz-asl/rotors_simulator): an existing ROS package (developed by ETH Zurich) which aims to simulate the behavior of aerial platforms (and also sensors) when control algorithms are in the loop.

At the time of development - I know that something happened in the middle, but nothing official - RotorS was available only for the Indigo Igloo and Kinetic Kame distros of ROS. Therefore, we decided to develop our contribution for these two software releases of ROS.

However, apart from the history behind the package, starting from your question I found [open issues on the RotorS GitHub repo](https://github.com/ethz-asl/rotors_simulator/issues?utf8=%E2%9C%93&q=is%3Aissue+melodic). It seems that they are going ahead, but I'm not sure. In the meantime, I suggest you downgrade your ROS distro and starting with Kinetic. 

I opened a milestone and branch on CrazyS, as well. I linked them to a pull request and will examine the possibility of adapting CrazyS to the Melodic distro of ROS.

I hope to reply you soon.

antbono, 2019-12-09:
It would be great to have this package available with ros melodic and gazebo 9!!! More realistic simulation setups are now possibile with the latest tools of matlab/simulink :)

gsilano, 2019-12-10:
The package is already compatible with Gazebo 9. You can install the 3D environment on ROS Kinetic by following [this guide](https://medium.com/@abhiksingla10/setting-up-ros-kinetic-and-gazebo-8-or-9-70f2231af21a).

antbono, 2019-12-10:
Thank you for this info! I'm going to try it :)

gsilano, 2019-12-12:
I'm working on this issue. I opened a proper branch (feature/gazebo9) for containing the code.

gsilano, 2019-12-13:
Hi @antbono!! I have almost completed the migration from Gazebo 7 to 9. As soon as the last tests have been completed, I will commit the code on GitHub. Would you like to be my beta tester? I would like to understand if I missed something in the list of commands to run RotorS, and therefore CrazyS, with Gazebo 9. Thanks!!

antbono, 2019-12-14:
Yes for sure! :D I can't promise anything about the timing, but I'll definitely want to try it! :D
(same holds for the BebopS package :)

gsilano, 2019-12-28:
Hi @AKR2470!! I just uploaded the instructions in the README.md to use the package with ROS Melodic and Gazebo 9.

@antbono, I have tested the package and fixed the issues you described. Now, the CrazyS is full compatible with Gazebo 9 and ROS Kinetic Kame. If you have time, don't hesitate to try my guide and highlight typos and / or mistakes.

So, I will close this issue as resolved. Please feel free to open another if problems appear.

closed 2019-12-28 by gsilano

AKR2470, 2019-12-29:
@gsilano Thank you! I'll surely checkout.

antbono, 2020-01-14:
Hi everyone! 
I am very happy that @gsilano did these new improvements to the package. Thank you very much!
I have just tried to follow the instruction for using CrazyS with gazebo9 on ubuntu 16.04. There are a few typos in the _Ubuntu 16.04 with ROS Kinetic and Gazebo 7_ paragraph.
In the first step: 
- correct the last command. the package 'pythonrosinstall-generator' should be 'python-rosinstall-generator'

In the second step:
- after the 'catkin init' command you should execute 'cd ~/catkin_ws/src'

In the _Ubuntu 16.04 with ROS Kinetic and Gazebo 9_ paragraph the bash script `gazebo9.sh` has a typo. Instead of `~/` you should use `./`. You can find a corrected version attached.

[gazebo9.txt](https://github.com/gsilano/CrazyS/files/4059929/gazebo9.txt)

Hope to hear from you soon! :D