gsilano/CrazyS #17: publishing waypoint

melotus, 2019-10-30:
Im trying to publish waypoints for crazyflie using the program.
[https://paste.ubuntu.com/p/7TsgQKv4vH/](url)

the hovering part is working properly and its not reaching the next point.
pls if someone could help me.

gsilano, 2019-10-30:
Hi! The link you provided does not work. Please, could you check?
Anyway, as discussed in other issues and in the reference papers, the hovering example proposed in this repo is not a trajectory controller. It is just an example of how you can control the nano-quadrotor using signals come from the virtual odometry sensor and the on-board IMU (it also has been simulated, take a look a the publications section in the Wiki)

melotus, 2019-10-30:
pl check this link

https://paste.ubuntu.com/p/7TsgQKv4vH/

gsilano, 2019-10-30:
I got the file but, as it is written in the comment, we developed a position and not a trajectory controller. There is any node that takes care of generating a suitable trajectory to follow starting from a sequence of waypoints.

gsilano, 2019-12-28:
In one of the next versions of the software, we will add a trajectory tracking controller. As a reminder, I will keep this problem open and then close it when the job is done.

gsilano, 2020-02-23:
#27, #28, and #29 should fix this issue. Stay tuned on the PRs.

gsilano, 2020-04-23:
Hey bot! I'm still working on this. Remove the `wontfix` tag.

gsilano, 2022-04-16:
Closed for inactivity. Add a reminder in #86 for this.

closed 2022-04-16 by gsilano