gsilano/CrazyS #18: Wiki Interfacing CrazyS through MATLAB --- Custom messages

antbono, 2019-12-06:
I suggest @gsilano to update the the last part of the wiki page about interfacing with matlab. The link you provide to use custom messages on windows without building them, actually doesn't have the jar files. In addition to provide them, you could use the suggestions provided in https://www.mathworks.com/help/ros/ug/ros-custom-message-support.html
to update the last part of wiki. I hope this could help.

gsilano, 2019-12-07:
HI @antbono!! Thanks for having opened an issue on CrazyS. I will check and let you know ASAP. I will edit this message. Thus, stay tuned on this discussion.

gsilano, 2019-12-28:
HI @antbono!! I just fixed the issue by adding the missing files. Thanks for the help you have given me to improve CrazyS

closed 2019-12-28 by gsilano