gsilano/CrazyS #19: Wiki Interfacing CrazyS through MATLAB --- CrazyflieController_RST

antbono, 2019-12-06:
I write this post to warn other users and @gsilano about some bugs with the simulink implementation of the crazyflie controller.

1. In the wiki a folder with a matlab script and a simulink model is provided. For some mismatches in the variables names between these two files you can't use them. Here you can find a corrected version of the simulink model.
[CrazyflieController_RST.zip](https://github.com/gsilano/CrazyS/files/3934357/CrazyflieController_RST.zip)

2. When launch the  crazyflie2_without_controller.launch you must set the argument "pause:=false" otherwise the matlab script won't work.

gsilano, 2019-12-07:
Hi @antbono!! Thanks for opening an issue. I will check and give it back to you ASAP.

gsilano, 2019-12-28:
@antbono, thanks for your help. I just added your tips to the Wiki (CrazyS, BebopS, and RotorS) and entered your name on the contribution list.

closed 2019-12-28 by gsilano

antbono, 2020-01-14:
You are very welcome. It was not necessary to add me to the list for so little help. Anyway thank you for the kind thought :)

Fjbarchesky, 2020-03-06:
Hello @gsilano, I am having some trouble with the supplied launch file in this MATLAB tutorial. After running the crazyflie2_without_controller.launch (minor change to file to be, paused = false and verbose = true), I receive no errors/warnings. The MAV does not appear in Gazebo, is this normal? I am able to successfully launch the other example in CrazyS (roslaunch rotors_gazebo crazyflie2_hovering_example.launch) without any problems. I am using Ubuntu 18.04 / Melodic / Gazebo 9 per the tutorial in CrazyS.

I am using the command below to execute the launch: 

`roslaunch -p 1234 ~/catkin_crazys_ws/src/CrazyS/rotors_gazebo/launch/crazyflie2_without_controller.launch`

If I am correct, I am forcing the master port to be 1234 with this command, and in the 'main.m' I changed line #13 (since its both my machine running Gazebo and ROS) to be my machines URL and port 1234. I was required to comment/un-comment the supplied 'main.m' to be have the lines #17 commented, line #18 un-commented, line #19 commented, and line #20 un-commented. (I believe this is due to the structure of the custom messages that was required to be added.)

Therefore that section of code looks like this: 

`%Topics will be used during the simulation
% sub = rossubscriber('/crazyflie2/odometry_sensor1/odometry'); % contains the drone state
sub = rossubscriber('/crazyflie2/odometry_sensor1/odometry','nav_msgs/Odometry'); % contains the drone state
% sub2 = rossubscriber('/crazyflie2/motor_speed'); 
sub2 = rossubscriber('/crazyflie2/motor_speed','mav_msgs/Actuators'); 
[pubCmd, msgCmd] = rospublisher('/crazyflie2/command/motor_speed','mav_msgs/Actuators');
`
The MATLAB 'main.m' script hangs/pauses at line #188 which is this command: 

`msgS  = receive(sub);`

I believe this is fairly expected since the Gazebo appears to not have launched correctly and the MATLAB cannot talk to Gazebo. 

Thanks for your help in advance.

gsilano, 2020-03-06:
Hi @Fjbarchesky! Since your problem does not fit well with this issue, I think it would be better to open a new one. Thanks!

Fjbarchesky, 2020-03-06:
Oops didn't mean to delete, created a new issue here: #36