gsilano/CrazyS #24: Ubuntu 16.04 with ROS Kinetic and Gazebo 9 --- crazyflie2_hovering_example.launch

antbono, 2020-01-14:
Hi everyone! 
I have just followed the instructions for installing CrazyS with the gazebo9 and ubuntu 16.04 (please see [this issue for possible corrections)](https://github.com/gsilano/CrazyS/issues/23) but I can't execute the basic usage example of the readme file. This error appears:
`You must call ros::init() before creating the first NodeHandle`
I googled it but i couldn't find an easy solution. 
@gsilano did you have the same problem when testing this installation?
Thank you for your collaboration

[image]

gsilano, 2020-01-14:
This problem appears when there is an error with the `bashrc` file, maybe you forgot to source it, or with the `rosdep` command. I had it and I fixed it in this way. Try to run `roscore` and see if everything is ok.
Then, let me know.

antbono, 2020-01-14:
`roscore` runs without errors for me...

gsilano, 2020-01-15:
It's strange to me. Have you tried running Gazebo?  Is the 9.11 Gazebo release working? Other problems might be related to missing files. I followed the same procedure on different machines, and it worked to me.

antbono, 2020-01-15:
I have just done a new virtual machine with a fresh ubuntu 16.04 but the same error occurred. I followed the instructions listed in the _Ubuntu 16.04 with ROS Kinetic and Gazebo 7_ and _Ubuntu 16.04 with ROS Kinetic and Gazebo 9_ with a few exceptions: 
-  The first two commands of the _Ubuntu 16.04 with ROS Kinetic and Gazebo 7_ guide didn't work for me. The first instruction because there should be `'` instead of `. The second didn't work so I used the key used in the official [ros-kinetic installation page](http://wiki.ros.org/kinetic/Installation/Ubuntu)

- The gazebo.sh arises some exceptions because sometimes you try to copy folders without `-r` option. I have attached a simpler version. [gazebo9.txt](https://github.com/gsilano/CrazyS/files/4065750/gazebo9.txt)

Both the command `roscore`  and `gazebo` work and the actual version of gazebo is 9.11. I haven't tried a to use gazebo with other simulations yet because the most I have ready work only with gazebo7. I update you if I can run something.
Thank you again

gsilano, 2020-01-16:
Update the `README.md` file basing on your comments. I tried the `gazebo9.sh` file and it worked for me. I really don't understand why. However, your issue may help people who have the same problem as yours.

About `libgazebo`, again it seems strange to me because I have not encountered any errors of this type. However, as I can understand, everything works well. Therefore, I can close this problem. I'm right?

antbono, 2020-01-16:
I am very sorry but my last update was misleading (I have just corrected it). Unfortunately the same error occurs all the times. I have done different test I can't understand what is going wrong... Have you tried with a fresh new ubuntu16 machine? Thank you again for you attention

gsilano, 2020-01-17:
Dear @antbono, I really do not understand what the problem is. So, today I created two virtual machines starting from scratch and everything was fine. Here are the links to download them. I hope they will be useful to you. I will also include these machines in the Wiki and in the README.md file.

[Ubuntu 16.04 with ROS Kinetic and Gazebo 7](https://mega.nz/#!oxdwzaIY!C27ekXwbObbiU37BrVQSjeRMP2HCACVgQj1WbvVmXlA)
[Ubuntu 16.04 with ROS Kinetic and Gazebo 9](https://mega.nz/#!550ASCrZ!JYm1mFFlYa9fNh_Cb4Z6p7l9k6Vu5tpRJB3NjMoSF1A)

To run the virtual machines you can use Oracle VirtualBox. Please, consider that you can improve the performances allocating much more RAM, GPU, and CPU.

```
USER: user
PASS: password
Keyboard layout: Italian
Language: English
```

antbono, 2020-01-17:
Thank you very much @gsilano. I have just tried your machine with gazebo9 and everything works. I'll try to figure out what are the differences between this machine and mine. If you agree, we can leave this topic open for some other days: maybe someone else meets my same problem.

closed 2020-01-18 by gsilano