gsilano/CrazyS #25: Ubuntu 18.04 with ROS Melodic and Gazebo 9 --- hovering example not working

antbono, 2020-01-17:
I have just installed ubuntu 18.04 and followed the instructions reported in the readme file. All the steps run successfully and I can run the hovering example. The problem is that the crazyflie doesn't move at all. Nothing happens in the simulation. I tried to apply some forces to the crazyflie to see some changes in the rqt_plots but nothing happens. The rqt_graph at the moment of the simulation doesn't show any incoming connection to gazebo node. 
Thank you for your attention

[image]
[image]

gsilano, 2020-01-17:
Did you wait five seconds for the activation of the control algorithm? Here what is reported in the README.md file.

> The visual outcome will see the nano-quadcopter taking off after 5s (time after which the hovering example node publishes the trajectory to follow) and flying one meter above the ground, at the same time keeping near to zero the position components along x and y-axis.

antbono, 2020-01-17:
Yes I waited. The attached gazebo screenshot reports 1.31 minutes

gsilano, 2020-01-18:
As I can read from the log, two are the main errors:

* The `enable_state_estimator` variable is equal to a strange number. It should be a boolean value, zero or one.
* The Crazyflie controller does not receive info one the drone state. I mean, any message comes from the Odometry sensor.

Now, I will take a look at the code, but these should be problems that I fixed a long time ago with the latest software release.

chjz1024, 2020-01-18:
I encountered the same problem running the hovering example. It was still okay when I tried that about 2 weeks ago (sorry that I forgot the exact time). I then reinstalled it today and it doesn't work now. The same error occurs when I tried running the original rotorS hovering example. This is my log as a reference

> chjz@hpc:~/Desktop/catkin_workspaces/catkin_ws_crazyS$ roslaunch rotors_gazebo crazyflie2_hovering_example.launch
> ... logging to /home/chjz/.ros/log/7fe35d00-39c9-11ea-9451-a85e455aff8e/roslaunch-hpc-28626.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> xacro: in-order processing became default in ROS Melodic. You can drop the option.
> started roslaunch server http://hpc:33697/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /crazyflie2/position_controller_node/attitude_gain_ki/phi: 0.0349
>  * /crazyflie2/position_controller_node/attitude_gain_ki/theta: 0.0349
>  * /crazyflie2/position_controller_node/attitude_gain_kp/phi: 0.0611
>  * /crazyflie2/position_controller_node/attitude_gain_kp/theta: 0.0611
>  * /crazyflie2/position_controller_node/enable_state_estimator: False
>  * /crazyflie2/position_controller_node/hovering_gain_kd/z: 373
>  * /crazyflie2/position_controller_node/hovering_gain_ki/z: 3.15
>  * /crazyflie2/position_controller_node/hovering_gain_kp/z: 70
>  * /crazyflie2/position_controller_node/inertia/xx: 1.657171e-05
>  * /crazyflie2/position_controller_node/inertia/xy: 0.0
>  * /crazyflie2/position_controller_node/inertia/xz: 0.0
>  * /crazyflie2/position_controller_node/inertia/yy: 1.657171e-05
>  * /crazyflie2/position_controller_node/inertia/yz: 0.0
>  * /crazyflie2/position_controller_node/inertia/zz: 2.9261652e-05
>  * /crazyflie2/position_controller_node/mass: 0.027
>  * /crazyflie2/position_controller_node/rate_gain_ki/p: 0
>  * /crazyflie2/position_controller_node/rate_gain_ki/q: 0
>  * /crazyflie2/position_controller_node/rate_gain_ki/r: 95.6839
>  * /crazyflie2/position_controller_node/rate_gain_kp/p: 1000
>  * /crazyflie2/position_controller_node/rate_gain_kp/q: 1000
>  * /crazyflie2/position_controller_node/rate_gain_kp/r: 1000
>  * /crazyflie2/position_controller_node/rotor_configuration/0/angle: -3.14159265359
>  * /crazyflie2/position_controller_node/rotor_configuration/0/arm_length: 0.046
>  * /crazyflie2/position_controller_node/rotor_configuration/0/direction: -1.0
>  * /crazyflie2/position_controller_node/rotor_configuration/0/rotor_force_constant: 1.28192e-08
>  * /crazyflie2/position_controller_node/rotor_configuration/0/rotor_moment_constant: 0.005964552
>  * /crazyflie2/position_controller_node/rotor_configuration/1/angle: 1.57079632679
>  * /crazyflie2/position_controller_node/rotor_configuration/1/arm_length: 0.046
>  * /crazyflie2/position_controller_node/rotor_configuration/1/direction: 1.0
>  * /crazyflie2/position_controller_node/rotor_configuration/1/rotor_force_constant: 1.28192e-08
>  * /crazyflie2/position_controller_node/rotor_configuration/1/rotor_moment_constant: 0.005964552
>  * /crazyflie2/position_controller_node/rotor_configuration/2/angle: 3.14159265359
>  * /crazyflie2/position_controller_node/rotor_configuration/2/arm_length: 0.046
>  * /crazyflie2/position_controller_node/rotor_configuration/2/direction: -1.0
>  * /crazyflie2/position_controller_node/rotor_configuration/2/rotor_force_constant: 1.28192e-08
>  * /crazyflie2/position_controller_node/rotor_configuration/2/rotor_moment_constant: 0.005964552
>  * /crazyflie2/position_controller_node/rotor_configuration/3/angle: -1.57079632679
>  * /crazyflie2/position_controller_node/rotor_configuration/3/arm_length: 0.046
>  * /crazyflie2/position_controller_node/rotor_configuration/3/direction: 1.0
>  * /crazyflie2/position_controller_node/rotor_configuration/3/rotor_force_constant: 1.28192e-08
>  * /crazyflie2/position_controller_node/rotor_configuration/3/rotor_moment_constant: 0.005964552
>  * /crazyflie2/position_controller_node/xy_gain_ki/x: 5.72958
>  * /crazyflie2/position_controller_node/xy_gain_ki/y: -5.72958
>  * /crazyflie2/position_controller_node/xy_gain_kp/x: 3.594
>  * /crazyflie2/position_controller_node/xy_gain_kp/y: -3.594
>  * /crazyflie2/position_controller_node/yaw_gain_ki/yaw: 0
>  * /crazyflie2/position_controller_node/yaw_gain_kp/yaw: 0.0914
>  * /crazyflie2/robot_description: <?xml version="1....
>  * /crazyflie2/tf_prefix: 
>  * /gazebo/enable_ros_network: True
>  * /rosdistro: melodic
>  * /rosversion: 1.14.3
>  * /use_sim_time: True
> 
> NODES
[... 57 lines cut ...]
> [INFO] [1579334821.952806, 0.112000]: Spawn status: SpawnModel: Successfully spawned entity
> [crazyflie2/spawn_crazyflie2-4] process has finished cleanly
> log file: /home/chjz/.ros/log/7fe35d00-39c9-11ea-9451-a85e455aff8e/crazyflie2-spawn_crazyflie2-4*.log
> [ INFO] [1579334827.355271124, 5.116000000]: Publishing waypoint on namespace /crazyflie2: [0.000000, 0.000000, 1.000000].
> [ INFO] [1579334827.355686232, 5.117000000]: PositionController got first MultiDOFJointTrajectory message.
> [crazyflie2/hovering_example-6] process has finished cleanly
> log file: /home/chjz/.ros/log/7fe35d00-39c9-11ea-9451-a85e455aff8e/crazyflie2-hovering_example-6*.log
> Node::Advertise(): Error advertising a topic. Did you forget to start the discovery service?
> ^C[crazyflie2/orientation_plot-11] killing on exit
> [crazyflie2/position_plot-10] killing on exit
[... cut ...]
> [crazyflie2/position_controller_node-5] killing on exit
> [gazebo_gui-3] killing on exit
> [gazebo-2] killing on exit
> terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
>   what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
> [gazebo_gui-3] escalating to SIGTERM
> [rosout-1] killing on exit
> [master] killing on exit
> shutting down processing monitor...
> ... shutting down processing monitor complete
> done
>

gsilano, 2020-01-18:
I fixed the issue with the `enable_state_estimator` variable but, by mistake, I referred the wrong issue (#24 instead of this, #25). Anyway, the code is ok. I did not find any problem. I have not encountered any problems. Now, I'll check if there is a typo in the installation instructions.

chjz1024, 2020-01-18:
@gsilano Unfortunately the most recent commit still doesn't work. If it's not because of this package, then the problem might come from the original repositories like [rotors-simulator](https://github.com/ethz-asl/rotors_simulator), [mav_control_rw](https://github.com/ethz-asl/mav_control_rw) and [voxblox](https://github.com/ethz-asl/voxblox), cause that's the only big difference I've made since the previous installation. A virtual machine with ROS Melodic and Gazebo 9 configuration is also of great help if available. I'm also considering creating a docker image for the same job with gzclient running on the host machine, is that possible?

chjz1024, 2020-01-18:
Hi @gsilano , I did a fresh install of crazyS package in a new ros:melodic container docker but it still didn't work. It said some the `libmav_msgs.so` file is not found. I'm creating a virtual machine to verify this.

gsilano, 2020-01-18:
To understand the problem, I also created a fresh installation of ROS Melodic on Ubuntu 18.04. I think I discovered the problem: some Gazebo packages are missing in the default installation. In fact, following the instructions in the repository, I had the same problem illustrated by @antbono. So, I run the command mentioned below and everything was fine.

```console
$ sudo apt-get remove ros-melodic-gazebo* gazebo*
$ sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
$ wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
$ sudo apt-get update
$ sudo apt-get install gazebo9 gazebo9-* ros-melodic-gazebo-*
$ sudo apt upgrade
```

Please, try to use them and give me back. In the meantime, I will update the README.md file.

gsilano, 2020-01-18:
> @gsilano Unfortunately the most recent commit still doesn't work. If it's not because of this package, then the problem might come from the original repositories like [rotors-simulator](https://github.com/ethz-asl/rotors_simulator), [mav_control_rw](https://github.com/ethz-asl/mav_control_rw) and [voxblox](https://github.com/ethz-asl/voxblox), cause that's the only big difference I've made since the previous installation. A virtual machine with ROS Melodic and Gazebo 9 configuration is also of great help if available. I've also tried running a seperate `gzserver` inside docker container and `gzclient` in host machine and it works. The only problem is that you need to care about the path because the absolute path is generated inside the container.

I think this is a particular use case, so I cannot help you in defining the problem.

antbono, 2020-01-18:
@gsilano I followed the new readme and finally the simulation works! Thank you so much for your work :D

closed 2020-01-18 by gsilano

fabrizzio-gz, 2021-11-15:
> To understand the problem, I also created a fresh installation of ROS Melodic on Ubuntu 18.04. I think I discovered the problem: some Gazebo packages are missing in the default installation. In fact, following the instructions in the repository, I had the same problem illustrated by @antbono. So, I run the command mentioned below and everything was fine.
> 
> ```
> $ sudo apt-get remove ros-melodic-gazebo* gazebo*
> $ sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
> $ wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
> $ sudo apt-get update
> $ sudo apt-get install gazebo9 gazebo9-* ros-melodic-gazebo-*
> $ sudo apt upgrade
> ```
> 
> Please, try to use them and give me back. In the meantime, I will update the README.md file.

I followed these instructions to have the simulation run on Gazebo 9 (along with Ubuntu 18.04 and ROS Melodic). However, now, when I run `catkin build`, I get the following error message: 
```
Starting  >>> rotors_gazebo_plugins                                            
_______________________________________________________________________________
Errors     << rotors_gazebo_plugins:make /home/user/catkin_ws/logs/rotors_gazebo_plugins/build.make.007.log
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libignition-common1.so.1.0.1', needed by '/home/user/catkin_ws/devel/.private/rotors_gazebo_plugins/lib/librotors_gazebo_mavlink_interface.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
In file included from /usr/include/gazebo-9/gazebo/common/common.hh:8:0,
                 from /usr/include/gazebo-9/gazebo/gazebo_core.hh:19,
                 from /usr/include/gazebo-9/gazebo/gazebo.hh:20,
                 from /home/user/catkin_ws/src/CrazyS/rotors_gazebo_plugins/include/rotors_gazebo_plugins/gazebo_mavlink_interface.h:47,
                 from /home/user/catkin_ws/src/CrazyS/rotors_gazebo_plugins/src/gazebo_mavlink_interface.cpp:22:
/usr/include/gazebo-9/gazebo/common/Battery.hh:25:10: fatal error: sdf/sdf.hh: No such file or directory
 #include "sdf/sdf.hh"
          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/rotors_gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o] Error 1
make[1]: *** [CMakeFiles/rotors_gazebo_mavlink_interface.dir/all] Error 2
make: *** [all] Error 2
cd /home/user/catkin_ws/build/rotors_gazebo_plugins; catkin build --get-env rotors_gazebo_plugins | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << rotors_gazebo_plugins:make            [ Exited with code 2 ]     
```
I was able to run `catkin build` without any errors before updating the Gazebo packages.

gsilano, 2021-11-17:
Hi! Try to run the following steps, and then give me back. 

1. `catkin clean`. This will delete the previous build.
2. Follow the instructions in the `README.md` file again. Please don't skip any of the steps. A package might not have been installed correctly.
3. In case of `cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -u -e ode` (see #40), run this command `cp build/rotors_gazebo_plugins/libmav_msgs.so devel/lib/`.

fabrizzio-gz, 2021-11-17:
> Hi! Try to run the following steps, and then give me back.
> 
>     1. `catkin clean`. This will delete the previous build.
> 
>     2. Follow the instructions in the `README.md` file again. Please don't skip any of the steps. A package might not have been installed correctly.
> 
>     3. In case of `cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -u -e ode` (see [Arch Linux with ROS Melodic and Gazebo 10 -- no movement #40](https://github.com/gsilano/CrazyS/issues/40)), run this command `cp build/rotors_gazebo_plugins/libmav_msgs.so devel/lib/`.

Hi, thanks for the reply! 

I have followed the instructions of the README file again. I'm still getting the same problem. The simulation runs on Gazebo 9, but when I run `catkin build`, I get the following error message:

```
$ catkin build
...
Starting  >>> rotors_gazebo_plugins                                                                                                         
____________________________________________________________________________________________________________________________________________
Errors     << rotors_gazebo_plugins:make /home/user/catkin_ws/logs/rotors_gazebo_plugins/build.make.001.log                                
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libignition-common1.so.1.0.1', needed by '/home/user/catkin_ws/devel/.private/rotors_gazebo_plugins/lib/librotors_gazebo_mavlink_interface.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
In file included from /usr/include/gazebo-9/gazebo/common/common.hh:8:0,
                 from /usr/include/gazebo-9/gazebo/gazebo_core.hh:19,
                 from /usr/include/gazebo-9/gazebo/gazebo.hh:20,
                 from /home/user/catkin_ws/src/CrazyS/rotors_gazebo_plugins/include/rotors_gazebo_plugins/gazebo_mavlink_interface.h:47,
                 from /home/user/catkin_ws/src/CrazyS/rotors_gazebo_plugins/src/gazebo_mavlink_interface.cpp:22:
/usr/include/gazebo-9/gazebo/common/Battery.hh:25:10: fatal error: sdf/sdf.hh: No such file or directory
 #include "sdf/sdf.hh"
          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/rotors_gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o] Error 1
make[1]: *** [CMakeFiles/rotors_gazebo_mavlink_interface.dir/all] Error 2
make: *** [all] Error 2
cd /home/user/catkin_ws/build/rotors_gazebo_plugins; catkin build --get-env rotors_gazebo_plugins | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
............................................................................................................................................
Failed     << rotors_gazebo_plugins:make            [ Exited with code 2 ]                  
```
I have also realized that the shared library `libignition-common1.so.1.0.1` does not exist.
```
$ stat /usr/lib/x86_64-linux-gnu/libignition-common1.so.1.0.1
stat: cannot stat '/usr/lib/x86_64-linux-gnu/libignition-common1.so.1.0.1': No such file or directory
```
I believe that installing the missing Gazebo packages (with the commands below), installs `libignition 1.1.1` but `rotors_gazebo_plugin` still requires `libignition 1.0.1`.
> ```
> $ sudo apt-get remove ros-melodic-gazebo* gazebo*
> $ sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > [...]

fabrizzio-gz, 2021-11-17:
> Hi! Try to run the following steps, and then give me back.
>
>     1. `catkin clean`. This will delete the previous build.
>
>     2. Follow the instructions in the `README.md` file again. Please don't [...]

Hi, thank you for your reply!

After running `catkin clean` and following the `README.md` instructions again, my problem was solved. I was able to run `catkin build` with no error messages. 

Thanks again!