gsilano/CrazyS #26: co-simulation simulink gazebo --- help

antbono, 2020-01-29:
Hi guys! 
I created this issue because I can't use the tools of the matlab robotics toolbox which allows you to perform co-simulation between simulink and gazebo [here](https://www.mathworks.com/help/robotics/examples/perform-co-simulation-between-simulink-and-gazebo.html#PerformCoSimulationBetweenSimulinkAndGazeboExample-1).
I used the hovering example reported in the readme and, as suggested in the guide, I added the tag for the gazebo plugin in the basic_crazyflie.world file. This way I can establish the connection between gazebo and simulink but when I start the simulink simulation gazebo can't restart its time. I obtain the following error

[image]

Have anyone faced this problem?
Thank you!

gsilano, 2020-01-29:
I think you meant the `crazyflie_without_controller.launch` file, am I right? Could you provide the MATLAB/Simulink scripts you are using to establish a connection with the drone?

antbono, 2020-01-29:
Actually I used `crazyflie2_hovering_example.launch` because I simply wanted to read some data from the gazebo simulation. Attached you can find the modified version of `basic_crazyflie.world` and the simple simulink sheet I used. 

[files.zip](https://github.com/gsilano/CrazyS/files/4129141/files.zip)

gsilano, 2020-01-29:
The `crazyflie2_hovering_example.launch` file has not been developed to be used with MATLAB. Try to do the same with the file I suggested you, and then give me back with the feedback you receive from the environment.

antbono, 2020-01-29:
You are right! :D using `crazyflie_without_controller.launch` the cosimulation works. I tried with the hovering example because in the matlab guide they are able to restart a gazebo simulation already started via the simulink sheet. 
Do you think it is possible to use the simulink block attached in this message? 

[image]

I read the list of the /gazebo topics during the simulation but it seems that there is nothing related to the drone position.

antbono, 2020-01-29:
I have just tried to apply some forces to the drone as reported in the [matlab guide](https://www.mathworks.com/help/robotics/examples/perform-co-simulation-between-simulink-and-gazebo.html#PerformCoSimulationBetweenSimulinkAndGazeboExample-1) but it doesn't work. I have no error messages in the console. Attached you can find the simulink model used.
 
[testForces.zip](https://github.com/gsilano/CrazyS/files/4129441/testForces.zip)

gsilano, 2020-01-29:
> You are right! :D using `crazyflie_without_controller.launch` the cosimulation works. I tried with the hovering example because in the matlab guide they are able to restart a gazebo simulation already started via the simulink sheet.
> Do you think it is possible to use the simulink block attached in this message?
> 
> [image]
> 
> I read the list of the /gazebo topics during the simulation but it seems that there is nothing related to the drone position
> 
> /gazebo/link_states
> /gazebo/model_states
> /gazebo/set_link_state
> /gazebo/set_model_state

To tell the truth, I don't know. I did never worked on this. We should go into the code to understand what is wrong. I think that the MathWorks example works with a 3D rigid object and not a drone.

antbono, 2020-01-29:
I don't think the problem is the object to handle... In this other [guide](https://www.mathworks.com/help/robotics/examples/control-a-differential-drive-robot-in-simulink-and-gazebo.html) they apply two torques to the wheels rotational joints of a differential drive mobile robot. I think that maybe launching a crazyflie gazebo simulation without using any ros module (just like in the guides) could solve some conflicts. What do you think?

gsilano, 2020-01-31:
As I understand, from the files you attached, the CoSimulation Gazebo plugin allows you to apply forces and torques to the robot. While the remainder of the code describes the geometry, the collision model (it may be different from the robot geometry) and the joints (i.e., revolute, friction, etc.).

Now, if you want to do the same with the Crazyflie, we should understand how to add this plugin to the model. In fact, as you can see, it is written in a different format. Perhaps, you can add it as a sensor, but first I should understand what you would like to simulate, the aim, and so on.

So, as you can understand, it takes time, more to reason than to write the code. If your is a research project, we could start a collaboration on this. Please let me know, and feel free to write an e-mail.

antbono, 2020-02-03:
Hi Giuseppe! Thank you again for your attention. It would be a pleasure for me starting a new collaboration! :D 
My goal is to simulate some forces on the drone while it is using a particular control system. I tried to use your interface between [matlab and gazebo](https://github.com/gsilano/CrazyS/wiki/Interfacing-CrazyS-through-MATLAB) but unfortunately there are some limitations on the choice of the sampling time (we may talk about this in another topic). This new simulink tool, instead, let you choose whatever sampling time and applying a force is far easier than calling a gazebo service. 
I think that if you are able to provide me a urdf complete model of crazyflie and bebop, with all the tags for gazebo and the mesh files, maybe I can try to convert them in the sdf format and add in that file the gazebo_simulink plugin. Do you think it could work?

gsilano, 2020-02-03:
@antbono, let's discuss it on Skype, and then I will update the issue with the code and instructions. I would like to better understand how your control algorithm works and how I can implement what you need. Please send me an e-mail with you your account so that we can schedule a meeting.

gsilano, 2020-02-23:
I close this issue because, after a brief discussions on Skype, we understood that what @antbono suggested concerns a new feature, partially developed already within the package. The idea is to use the functionalities offered by the Robotics System Toolbox (find the link above) to synchronize MATLAB with Gazebo by using a parser without ROS in the loop. This requires designing new plugins or adapting the existing ones to provide to Simulink the data needed to control the drone dynamics.

I will keep track of this request for one of the next release.

closed 2020-02-23 by gsilano