gsilano/CrazyS #30: Joystick control for Crazyflie

naiveHobo, 2020-02-08:
The roll_pitch_yawrate_thrust_controller_node used by joystick control does not work for Crazyflie2 like it does for the other drone simulations available.

How should I proceed with setting up an environment where I'm able to control a Crazyflie using a joystick?

gsilano, 2020-02-09:
I took a look at the code. The actual Crazyflie control architecture does not allow to use the joystick. I will start working on this. Please, stay tuned on this issue for updates. 
I will also update the README.md file with instructions.

gsilano, 2020-02-09:
I've just completed the interface for all releases of CrazyS releases. Tomorrow I will test it and let you know. If it works, I will merge it into the master (and the proper branches) and update the README.md file with instructions on how to use it.

gsilano, 2020-02-10:
The new feature has been added to the repository. Thus, I close the issue.

naiveHobo, 2020-02-11:
Thrust commands work as expected but the roll and pitch commands are not working. The drone is only able to take flight and not tilt as per the roll and pitch commands.

gsilano, 2020-02-11:
What do you mean exactly? Did you try to change the gains in the `roll_pitch_yawrate_thrust_controller_crazyflie2.yaml` file? It may be a problem related to the on-board control architecture.

naiveHobo, 2020-02-12:
But the `roll_pitch_yaw_thrust_controller` is not using the on-board controller. It has a rate and attitude controller of it's own, right?

I didn't change the gains in the yaml file but even without that it should be able to work right?

When I send only roll/pitch commands with the joystick without any thrust, the corresponding 2 adjacent propellers should start rotating, but that does not happen.

gsilano, 2020-02-12:
Yes, the file uses the Crazyflie’s onboard controller. As for the real device, when you control the drone by using the joypad, the onboard controller manages to drive the motors.

It depends. What do you mean with right?

No! If you don’t apply any thrust, the propellers will never rotate. If you don’t thrust me, try with real device if you have it.

I didn’t get the last point. Could you be more clear?

naiveHobo, 2020-02-14:
The RollPitchYawrateThrustControllerCrazyflie gets it's odometry from the gazebo odometry plugin, right? The state of the drone is set on the basis of this odometry message which is received in the world frame, which means that the attitude and angular velocities of the drone are set in the world frame.
However, the PWM calculations are done to give angular velocities to the motor in the local coordinate system of the drone.

To give an example, if the drone was facing backwards such that the back motors are pointing in +x direction and the front motors are pointing in the -x direction, and a pitch command is given to tilt the drone towards the +x direction, the motor angular velocities will be calculated such that the back motors rotate faster than the front motors and the drone will actually tilt towards the -x direction. 

The odometry message will tell the drone that it is tilting towards the -x direction, but the pitch command received is still trying to achieve a +x tilt, so the PID controllers will keep increasing the delta_phi, delta_theta,  delta_psi commands but will never be able to reach the setpoint.

Basically, the joystick commands being received and the motor angular velocities being calculated are in the local frame of the drone and the odometry messages received are in the world frame, which would yield incorrect results.

Does this sound right to you?

gsilano, 2020-02-15:
Yes, you are right. The `RollPitchYawrateThrustControllerCrazyflie` gets the drone state from the odometry message. As you can see [from here](https://github.com/gsilano/CrazyS/blob/master/rotors_gazebo_plugins/src/gazebo_odometry_plugin.cpp), the `odometry plugin` expresses the linear and the angular velocities in body frame. If you don't thrust (please, check by your self because I could be wrong), try to rotate such measurements and see what happens.

What you are describing is modeled by the [equations reported](https://github.com/gsilano/CrazyS/blob/master/rotors_control/src/library/roll_pitch_yawrate_thrust_crazyflie.cpp#L121) in the `Control Mixer`. To better understand the control architecture, I suggest you read what is described in the [reference paper](https://giuseppesilano.net/publications/rosChapter19.pdf) (you find all references in the Wiki).

Maybe I did not understand well, but I don't totally agree with you. As you can see from the reference paper, tilting around the y-axis you generate a pitch angle (theta) and not w.r.t. x-axis. Also, the way in which the pitch angle increase should be different. If you don't trust (again, I could be wrong), create the map (PWM equations mentioned before) between the motors spinning and the command signals. The rows of the matrix are the motors velocities, while the columns are the command signals, i.e., thrust, roll, pitch, and yaw rate.

naiveHobo, 2020-02-15:
Yes, you're right! Sorry, I made that error about the pitch angle, I meant rate. Thanks for being so prompt with the answers, this is brilliant work!

closed 2020-02-15 by naiveHobo