gsilano/CrazyS #35: Missing dependencies in Joystick interface

naiveHobo, 2020-02-10:
The newly added joystick interface for crazyflie (fixing #30) uses mav_msgs::RollPitchYawrateThrustCrazyflie. This message type is also used in rotors_control/roll_pitch_yawrate_thrust_crazyflie.h.

I feel this is a typo and the message type which was supposed to be used was mav_msgs::RollPitchYawrateThrust.

gsilano, 2020-02-10:
Did you pull the latest changes in the `mav_comm` repository?

naiveHobo, 2020-02-11:
The latest version of mav_comm fixed the errors. However, there are still some problems with controlling a crazyflie using a joystick. Thrust commands work as expected but the roll and pitch commands are not working. The drone is only able to take flight and not tilt as per the roll and pitch commands.

closed 2020-02-11 by naiveHobo