gsilano/CrazyS #36: crazyflie2_without_controller.launch without MAV appearing

Fjbarchesky, 2020-03-06:
Hello @gsilano, I am having some trouble with the supplied launch file in this MATLAB tutorial. After running the crazyflie2_without_controller.launch (minor change to file to be, paused = false and verbose = true), I receive no errors/warnings. The MAV does not appear in Gazebo, is this normal? I am able to successfully launch the other example in CrazyS (roslaunch rotors_gazebo crazyflie2_hovering_example.launch) without any problems. I am using Ubuntu 18.04 / Melodic / Gazebo 9 per the tutorial in CrazyS.

I am using the command below to execute the launch:

roslaunch -p 1234 ~/catkin_crazys_ws/src/CrazyS/rotors_gazebo/launch/crazyflie2_without_controller.launch

If I am correct, I am forcing the master port to be 1234 with this command, and in the 'main.m' I changed line 13 (since its both my machine running Gazebo and ROS) to be my machines URL and port 1234. I was required to comment/un-comment the supplied 'main.m' to be have the lines 17 commented, line 18 un-commented, line 19 commented, and line 20 un-commented. (I believe this is due to the structure of the custom messages that was required to be added.)

Therefore that section of code looks like this:

%Topics will be used during the simulation
% sub = rossubscriber('/crazyflie2/odometry_sensor1/odometry'); % contains the drone state
sub = rossubscriber('/crazyflie2/odometry_sensor1/odometry','nav_msgs/Odometry'); % contains the drone state
% sub2 = rossubscriber('/crazyflie2/motor_speed');
sub2 = rossubscriber('/crazyflie2/motor_speed','mav_msgs/Actuators');
[pubCmd, msgCmd] = rospublisher('/crazyflie2/command/motor_speed','mav_msgs/Actuators');

The MATLAB 'main.m' script hangs/pauses at line 188 which is this command:

msgS = receive(sub);

I believe this is fairly expected since the Gazebo appears to not have launched correctly and the MATLAB cannot talk to Gazebo.

Thanks for your help in advance.

gsilano, 2020-03-06:
Ok, let's start from the beginning. I mean, let's start understanding why when you run the `roslaunch rotors_gazebo crazyflie2_without_controller.launch` command it does not work. Could you try to run the above command instead of the one you reported in the issue?

Then, copy and paste the verbose (if it does not work).

Fjbarchesky, 2020-03-06:
Well that appears to work, it comes up paused with the MAV off the ground.

Fjbarchesky, 2020-03-06:
So i think I understand this issue now, the text in the tutorial does not include a section of the launch file: 

```
    <group ns="$(arg mav_name)">
    <include file="$(find rotors_gazebo)/launch/spawn_mav_crazyflie.launch">
      <arg name="mav_name" value="$(arg mav_name)" />
      <arg name="model" value="$(find rotors_description)/urdf/mav_generic_odometry_sensor.gazebo" />
      <arg name="enable_logging" value="$(arg enable_logging)" />
      <arg name="enable_ground_truth" value="$(arg enable_ground_truth)" />
      <arg name="enable_state_estimator" value="$(arg enable_state_estimator)" />
      <arg name="log_file" value="$(arg log_file)"/>
    </include>

    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
    <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher" />
  </group>

```

gsilano, 2020-03-06:
Ok, great! It means that your configuration is ok. I mean, there are no problems with the ROS package. 

Now, the drone is waiting for commands, i.e., the propellers angular velocity. You can provide this command in open loop, typing on another terminal line the command

`rostopic pub /gazebo/command/motor_speed mav_msgs/Actuators '{angular_velocities: [1000, 1000, 1000, 1000]}'` 

or you can send it by using MATLAB together with the Robotic System Toolbox, as it is described in the [Wiki page](https://github.com/gsilano/CrazyS/wiki/Interfacing-CrazyS-through-MATLAB). Further information are described in the reference publications.

Fjbarchesky, 2020-03-06:
When I execute the following command in a terminal - Gazebo starts and the drone is visible: 
`roslaunch rotors_gazebo crazyflie2_without_controller.launch`
Simultaneously in another terminal I then enter the following command - nothing happens (excluding Sim Time increases): 
`rostopic pub /gazebo/command/motor_speed mav_msgs/Actuators '{angular_velocities: [1000, 1000, 1000, 1000]}'`

Side Note: I am able to execute the MATLAB 'main.m' file in addition with `roslaunch rotors_gazebo crazyflie2_without_controller.launch` the and the simulation runs through all 500 time steps and then eventually the drone crashes due to the MATLAB finishing its duration.

gsilano, 2020-03-07:
Ok! I think there is a typo in the name of the topic. I suggest you to the commands `rosnode info` and `rostopic info` to get the right name.

Of course, when the controller is disabled (MATLAB script ends) the drone falls down.

If you have other questions or doubts, please do not hesitate to ask. Otherwise, feel free to close the issue.

closed 2020-03-07 by Fjbarchesky