gsilano/CrazyS #37: DroneState.h is missing in rotors_gazebo/hovering_example_spline.h

Tonirayn, 2020-03-14:
Hi, 

I'm working with your repository for my master's thesis. I have to reproduce aggressive trajectory like flip with the crazyflie 2.1 and having a simulator like this is really useful. I know that you have not implemented yet the trajectory following but I notice that in the branch feature/controller-ros-melodic you have implemented the mellinger controller. However I try to compile it and it seems that a file is missing (mav_msgs/DroneState.h)

thanks for your work

gsilano, 2020-03-14:
Hi @Tonirayn!! First of all, I'm happy to know that you've found CrazyS useful for your research. So, as for the Mellinger's controller, this is part of an internship program between the University of Sannio and MIT. The work was mainly carried out by @rsonecha (https://www.mitathletics.com/sports/w-crewlt/2018-19/bios/sonecha_ria_toum). 
As you saw from PRs (#27, #28, and #29), I'm working to merge her work in CrazyS. Unfortunately, this is a very busy time for me and there is little free time to work on it.

Since you need it, I will try to do my best to merge at least the working part of the code. Please, stay tuned on the repository for new releases.

gsilano, 2020-04-22:
The code has just been made available. For more details see #27, #28, and #29.

closed 2020-04-22 by gsilano

MvB12, 2021-02-07:
hey,
Im working with same package and got same error..Ive downloaded newest version and checked with changed cmakelist too but I'm still getting same error..Along with Dronestate.h file, mav_msgs/RollPitchYawrateThrustCrazyflie.h is also missing..what should i do?

gsilano, 2021-02-08:
Could you provide me more details? Are you using the instructions reported in the `README.md` file? What are the Ubuntu and ROS releases?

MvB12, 2021-02-08:
yep..Ive followed as written...ive pasted the errors.Im using ubuntu 18, ros melodic, gazebo 9

/home/vishnu/agv_ws/src/CrazyS-master/rotors_joy_interface/include/rotors_joy_interface/joy_crazyflie.h:22:10: fatal error: mav_msgs/RollPitchYawrateThrustCrazyflie.h: No such file or directory
 #include <mav_msgs/RollPitchYawrateThrustCrazyflie.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

/home/vishnu/agv_ws/src/CrazyS-master/rotors_gazebo/include/rotors_gazebo/spline_trajectory_generator.h:39:11: fatal error: mav_msgs/DroneState.h: No such file or directory
  #include <mav_msgs/DroneState.h>
           ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

please let me know if more details are required?

gsilano, 2021-02-08:
This error usually comes up using the `catkin make` instead of the `catkin build` command. I simply suggest you to run `catkin clean`, and then `catkin build` again. Then, let me know.

MvB12, 2021-02-08:
hey, ive tried catkin build but it still throws exactly same errors..only those 2 packages are failing to build.

gsilano, 2021-02-08:
Could you show me the commands your run and also a screen of the terminal?

MvB12, 2021-02-08:
catkin build 

ive pasted the screen after catkin build

--------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/melodic
Workspace:                   /home/vishnu/agv_ws
--------------------------------------------------------
Build Space:        [exists] /home/vishnu/agv_ws/build
Devel Space:        [exists] /home/vishnu/agv_ws/devel
Install Space:      [unused] /home/vishnu/agv_ws/install
Log Space:          [exists] /home/vishnu/agv_ws/logs
Source Space:       [exists] /home/vishnu/agv_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------
[build] Found '26' packages in 0.0 seconds.                                    
[build] Package table is up to date.                                           
Starting  >>> codes                                                            
Starting  >>> crazyflie_controller                                             
Starting  >>> crazyflie_cpp                                                    
Starting  >>> crazyflie_demo                                                   
Finished  <<< crazyflie_cpp                         [ 0.8 seconds ]            
Starting  >>> crazyflie_description                                            
Finished  <<< codes                                 [ 0.9 seconds ]            
Starting  >>> libmavconn                                                       
Finished  <<< crazyflie_demo                        [ 0.8 seconds ]            
Starting  >>> mav_msgs                                                         
Finished  <<< crazyflie_controller                  [ 1.1 seconds ]            
Starting  >>> mav_state_machine_msgs                                           
Finished  <<< crazyflie_description                 [ 0.3 seconds ]            
Starting  >>> mav_system_msgs                                                  
Finished  <<< libmavconn                            [ 0.7 seconds ]            
Finished  <<< mav_state_machine_msgs                [ 1.1 seconds ]            
Starting  >>> mavros_msgs                                                      
Starting  >>> rotors_comm                                                      
Finished  <<< mav_msgs                              [ 1.6 seconds ]            
Starting  >>> rotors_description                                               
Finished  <<< rotors_description                    [ 0.3 seconds ]            
Finished  <<< mav_system_msgs                       [ 3.0 seconds ]            
Finished  <<< rotors_comm                           [ 2.1 seconds ]            
Starting  >>> rotors_evaluation                                                
Starting  >>> crazyflie_driver                                                 
Starting  >>> mav_planning_msgs                                                
Finished  <<< rotors_evaluation                     [ 0.6 seconds ]            
Starting  >>> rotors_control                                                   
Finished  <<< mavros_msgs                           [ 5.2 seconds ]            
Finished  <<< mav_planning_msgs                     [ 4.4 seconds ]            
Finished  <<< crazyflie_driver                      [ 6.8 seconds ]            
Starting  >>> rotors_joy_interface                                             
Starting  >>> mavros                                                           
_______________________________________________________________________________
Errors     << rotors_joy_interface:make /home/vishnu/agv_ws/logs/rotors_joy_interface/build.make.001.log
In file included from /home/vishnu/agv_ws/src/CrazyS-master/rotors_joy_interface/src/joy_crazyflie.cpp:18:0:
/home/vishnu/agv_ws/src/CrazyS-master/rotors_joy_interface/include/rotors_joy_interface/joy_crazyflie.h:22:10: fatal error: mav_msgs/RollPitchYawrateThrustCrazyflie.h: No such file or directory
 #include <mav_msgs/RollPitchYawrateThrustCrazyflie.h>
[... 71 lines cut ...]
/home/vishnu/agv_ws/src/CrazyS-master/rotors_control/src/library/roll_pitch_yawrate_thrust_crazyflie.cpp: At global scope:
/home/vishnu/agv_ws/src/CrazyS-master/rotors_control/src/library/roll_pitch_yawrate_thrust_crazyflie.cpp:224:21: error: ‘EigenRollPitchYawrateThrustCrazyflie’ in namespace ‘mav_msgs’ does not name a type
     const mav_msgs::EigenRollPitchYawrateThrustCrazyflie& roll_pitch_yawrate_thrust) {
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/vishnu/agv_ws/src/CrazyS-master/rotors_control/src/library/roll_pitch_yawrate_thrust_crazyflie.cpp: In member function ‘void rotors_control::RollPitchYawrateThrustControllerCrazyflie::SetRollPitchYawrateThrust(const int&)’:
/home/vishnu/agv_ws/src/CrazyS-master/rotors_control/src/library/roll_pitch_yawrate_thrust_crazyflie.cpp:226:3: error: ‘roll_pitch_yawrate_thrust_’ was not declared in this scope
[... cut ...]
Failed    <<< rotors_control                        [ 17.0 seconds ]           
[build] Summary: 16 of 23 packages succeeded.                                  
[build]   Ignored:   3 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: 5 packages were abandoned.                                
[build]   Failed:    2 packages failed.                                        
[build] Runtime: 25.0 seconds total.

MvB12, 2021-02-09:
hey, can i please know if i had done any mistake or any insight in error I'm getting

gsilano, 2021-02-10:
As you can see, the error originates from the `mav_msgs` package. Maybe, you are not using the right branch. Could you check that you are on the right one?

MvB12, 2021-02-10:
I'm using this package(https://github.com/ethz-asl/mav_comm/)...I downloaded the master branch of this..If this is not the right branch can u point me at the right one?

gsilano, 2021-02-10:
We found the issue. You need to clone and use this repository `https://github.com/gsilano/mav_comm.git`. Please, follow the instructions as detailed in the README.md file. Thanks!

gsilano, 2025-11-23:
That’s strange. Try cleaning your workspace. It often helps. Just run `catkin clean` and then `catkin make` again.