gsilano/CrazyS #38: Which parameter in gazebo will ultimately control uav motion?

Tcloser, 2020-03-22:
Hi,

I'm working with your repository for thesis too.
I calculated the motor speed by changing the lee_position_control.cpp control method and posted it on the topic of / uav / motor_speed. After that I could read the correct result in gazebo / command / motor_speed and It ’s normal except the plane doesn’t move.
So i want to know which parameter in gazebo will ultimately control uav motion?

Tcloser, 2020-03-22:
Sorry i solved it :P

closed 2020-03-22 by Tcloser

gsilano, 2020-03-22:
Hi @Tcloser! I'm happy to hear that you found CrazyS useful for your research activity. As for your question, could you be more specific? How can the Lee's position controller be used to control the system? I mean, which are the gains you use to control the drone? It would be nice if you could share these parameters with the whole community.

Tcloser, 2020-03-24:
I obtained the position, orientation, trajectory, motor speed and other data by extracting the data from the odom ,motor_speed topics. Then I built a network to learn the control method, after that t I combined the ros and tensorflow to use the neural network to calculate the output of the motor to control the uav.It's fun ,and we will open source it later.

gsilano, 2020-03-24:
Oh, great!! Please, do not forget to reference this issue in one of your commits or publish a message with the link to the repo. Thanks!!