gsilano/CrazyS #39: Multiranger

sanketgoyal, 2020-03-24:
Hello, I am trying to do things using the multi ranger and flow deck of crazyflie. Is there any source you could share which might be useful since I did not find any packages for these decks compatible with ROS.

gsilano, 2020-03-24:
CrazyS aims to simulate the dynamical behavior of the Crazyflie 2.0 nano-quadrotor by also integrating both low-level controller and on-board sensors. As it is described in the [reference papers](https://github.com/gsilano/CrazyS/wiki/Publications), the information comes from an odometry virtual sensor and a simulated IMU (Inertial Measurement Unit). 

The package does not provide any function to interface the simulator with the real device. To this, I suggest you take a look at the [crazyflie_mpc](https://github.com/bcbarbara/crazyflie_nmpc) ROS package.

sanketgoyal, 2020-04-07:
I checked out the repo but still was not able to find how to access multi ranger in ROS. Have you worked on that?

gsilano, 2020-04-07:
Sorry, I never worked with it. I can suggest taking a look at the [official website](https://www.bitcraze.io/products/multi-ranger-deck/). In case you discover something, please let us know.

closed 2020-04-22 by gsilano