gsilano/CrazyS #40: Arch Linux with ROS Melodic and Gazebo 10 -- no movement

KomaGR, 2020-03-25:
I am trying to make this work with this setup. To my knowledge I've installed all the dependencies and I've understood how message passing works in the hovering example.

Running the hovering example launches gazebo succesfully, messages look to be exchanged succesfully, but there is no movement from the drone after the waypoint is published. The drone simply remains motionless. Any pointers would be appreciated.

Here is  the [build.log](https://github.com/gsilano/CrazyS/files/4381853/build.log) from `catkin build`. There are also some warnings there about `empy` but I didn't think it'd be much of a problem.

Find below the log from the example, which looks OK minus a warning about a deprecated function. 

```
➜  catkin_wc roslaunch rotors_gazebo crazyflie2_hovering_example.launch
... logging to /home/koma/.ros/log/6a3842ae-6ead-11ea-892d-10c37b95af0c/roslaunch-TheMothership-26486.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
/opt/ros/melodic/lib/python3.8/site-packages/roslaunch/loader.py:412: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  data = yaml.load(text)
started roslaunch server http://TheMothership:42757/

SUMMARY
========

PARAMETERS
 * /crazyflie2/position_controller_node/attitude_gain_ki/phi: 0.0349
 * /crazyflie2/position_controller_node/attitude_gain_ki/theta: 0.0349
 * /crazyflie2/position_controller_node/attitude_gain_kp/phi: 0.0611
 * /crazyflie2/position_controller_node/attitude_gain_kp/theta: 0.0611
 * /crazyflie2/position_controller_node/csvFilesStoring: False
 * /crazyflie2/position_controller_node/csvFilesStoringTime: 15.0
 * /crazyflie2/position_controller_node/enable_state_estimator: False
 * /crazyflie2/position_controller_node/hovering_gain_kd/z: 373
 * /crazyflie2/position_controller_node/hovering_gain_ki/z: 3.15
 * /crazyflie2/position_controller_node/hovering_gain_kp/z: 70
 * /crazyflie2/position_controller_node/inertia/xx: 1.657171e-05
 * /crazyflie2/position_controller_node/inertia/xy: 0.0
 * /crazyflie2/position_controller_node/inertia/xz: 0.0
 * /crazyflie2/position_controller_node/inertia/yy: 1.657171e-05
 * /crazyflie2/position_controller_node/inertia/yz: 0.0
 * /crazyflie2/position_controller_node/inertia/zz: 2.9261652e-05
 * /crazyflie2/position_controller_node/mass: 0.027
 * /crazyflie2/position_controller_node/rate_gain_ki/p: 0
 * /crazyflie2/position_controller_node/rate_gain_ki/q: 0
 * /crazyflie2/position_controller_node/rate_gain_ki/r: 95.6839
 * /crazyflie2/position_controller_node/rate_gain_kp/p: 1000
 * /crazyflie2/position_controller_node/rate_gain_kp/q: 1000
 * /crazyflie2/position_controller_node/rate_gain_kp/r: 1000
 * /crazyflie2/position_controller_node/rotor_configuration/0/angle: -3.14159265359
 * /crazyflie2/position_controller_node/rotor_configuration/0/arm_length: 0.046
 * /crazyflie2/position_controller_node/rotor_configuration/0/direction: -1.0
 * /crazyflie2/position_controller_node/rotor_configuration/0/rotor_force_constant: 1.28192e-08
 * /crazyflie2/position_controller_node/rotor_configuration/0/rotor_moment_constant: 0.005964552
 * /crazyflie2/position_controller_node/rotor_configuration/1/angle: 1.57079632679
 * /crazyflie2/position_controller_node/rotor_configuration/1/arm_length: 0.046
 * /crazyflie2/position_controller_node/rotor_configuration/1/direction: 1.0
 * /crazyflie2/position_controller_node/rotor_configuration/1/rotor_force_constant: 1.28192e-08
 * /crazyflie2/position_controller_node/rotor_configuration/1/rotor_moment_constant: 0.005964552
 * /crazyflie2/position_controller_node/rotor_configuration/2/angle: 3.14159265359
 * /crazyflie2/position_controller_node/rotor_configuration/2/arm_length: 0.046
 * /crazyflie2/position_controller_node/rotor_configuration/2/direction: -1.0
 * /crazyflie2/position_controller_node/rotor_configuration/2/rotor_force_constant: 1.28192e-08
 * /crazyflie2/position_controller_node/rotor_configuration/2/rotor_moment_constant: 0.005964552
 * /crazyflie2/position_controller_node/rotor_configuration/3/angle: -1.57079632679
 * /crazyflie2/position_controller_node/rotor_configuration/3/arm_length: 0.046
 * /crazyflie2/position_controller_node/rotor_configuration/3/direction: 1.0
 * /crazyflie2/position_controller_node/rotor_configuration/3/rotor_force_constant: 1.28192e-08
 * /crazyflie2/position_controller_node/rotor_configuration/3/rotor_moment_constant: 0.005964552
 * /crazyflie2/position_controller_node/user_account: giuseppe
 * /crazyflie2/position_controller_node/xy_gain_ki/x: 5.72958
 * /crazyflie2/position_controller_node/xy_gain_ki/y: -5.72958
[... 56 lines cut ...]
[INFO] [1585150170.209848, 0.238000]: Spawn status: SpawnModel: Successfully spawned entity
[crazyflie2/spawn_crazyflie2-4] process has finished cleanly
log file: /home/koma/.ros/log/6a3842ae-6ead-11ea-892d-10c37b95af0c/crazyflie2-spawn_crazyflie2-4*.log
[ INFO] [1585150187.638864568, 15.123000000]: Stop waiting.
[ INFO] [1585150187.639955674, 15.124000000]: Publishing waypoint on namespace /crazyflie2: [0.000000, 0.000000, 1.000000].
[ INFO] [1585150187.640127342, 15.124000000]: PositionController got first MultiDOFJointTrajectory message.
[crazyflie2/hovering_example-6] process has finished cleanly
log file: /home/koma/.ros/log/6a3842ae-6ead-11ea-892d-10c37b95af0c/crazyflie2-hovering_example-6*.log

^C[crazyflie2/quaternion_to_rpy-9] killing on exit
[crazyflie2/joint_state_publisher-8] killing on exit
[crazyflie2/robot_state_publisher-7] killing on exit
[crazyflie2/position_controller_node-5] killing on exit
[gazebo-2] killing on exit
[gazebo_gui-3] killing on exit
terminate called after throwing an instance of 'boost::wrapexcept<boost::lock_error>'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[gazebo_gui-3] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

gsilano, 2020-03-25:
Hi @KomaGR! I've never tried to run the system with an OS distribution other than Ubuntu. However, if it builds this means that it is not a problem related to the code itself. 

Reading your message, it looks like a problem that I just faced in #25. I suggest to remove and then to install again all gazebo9 packages. Thus, run these commands

```console
$ sudo apt-get remove ros-melodic-gazebo* gazebo*
$ sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
$ wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
$ sudo apt-get update
$ sudo apt-get install gazebo9 gazebo9-* ros-melodic-gazebo-*
$ sudo apt upgrade
```

I'm pretty sure this is the problem because if you see the log file, no message appears regarding the odometry message. I mean, "PositionController got first odometry message."

Try, and give me back.

KomaGR, 2020-03-25:
Hi @gsilano, thanks for your swift reply. I've rebuilt all the packages but nothing changes. Could it be an incompatibility between gazebo9 and gazebo10?

Regardless, based on what you said about the messages appearing in the logs I started looking on the route the message must take to arrive to [the line you mentioned](https://github.com/gsilano/CrazyS/blob/59f227f4caddafbca0c2872427b7a5539f1c95a3/rotors_control/src/nodes/position_controller_node.cpp#L208). It can be verified by the logs that the message leaves the `hovering_example` on the `COMMAND_TRAJECTORY` topic and is received by  `MultiDofJointTrajectoryCallback()` [here](https://github.com/gsilano/CrazyS/blob/59f227f4caddafbca0c2872427b7a5539f1c95a3/rotors_control/src/nodes/position_controller_node.cpp#L75-L100). As you say, no odometry message. Yet.

**However:**
I do not understand who publishes on the `mav_msgs::default_topics::ODOMETRY` topic. I see multiple subscribers, `position_controller_node` is among them with the `OdometryCallback()`. But I can't seem to find who publishes at that topic. 

Any ideas on where I should be looking next for this would be appreciated.

gsilano, 2020-03-25:
It could be, it could be. I have never tested the package with Gazebo 10. When I read the issue, I realized that you were working with Melodic on Arch and that's it.
Why would you want to use Gazebo 10 instead of Gazebo 9? Is there a particular reason to use it?

The RotorS package used the basis for CrazyS is very complex and, unfortunately, there are no guides on how to use it correctly or how the software is structured. However, the `rotors_gazebo_plugins` folder contains all the plugins used for the motors, odometry sensor, etc. You could start from there.

If you find something interesting, please let me know. The problem may lie in the `CMakeList.txt` file because I fought hard to make the package compatible with Gazebo 9.

KomaGR, 2020-04-12:
Hi @gsilano. I've been intermittently trying to solve this issue, in order to use the simulation for some experiments. I have studied the source code of the involved nodes. I have some questions that would greatly help if answered:

### In what module does the robot read the commands?
I can see that the `PositionControllerNode::MultiDofJointTrajectoryCallback` is receiving a message (you can see it in the log above). I also noticed that in the Kinetic VM you provide, it is the same callback (and **not** the `PositionControllerNode::OdometryCallback`). 
A trajectory is calculated from that message and then pushed in a command queue:
https://github.com/gsilano/CrazyS/blob/59f227f4caddafbca0c2872427b7a5539f1c95a3/rotors_control/src/nodes/position_controller_node.cpp#L88-L94
The position controller sets trajectory point, but this only raises a boolean flag:
https://github.com/gsilano/CrazyS/blob/59f227f4caddafbca0c2872427b7a5539f1c95a3/rotors_control/src/library/position_controller.cpp#L250-L252

1. **Do you know where does the command go from here?**
2. **Is `PositionController::CallbackHightLevelControl` supposed to be running on a loop?**

------

> Why would you want to use Gazebo 10 instead of Gazebo 9? Is there a particular reason to use it?

For better or for worse, this is what my distro provides.

gsilano, 2020-04-12:
I don't know if I really understand what you are trying to say, so I will try to be as clear as possible. 

The position, i.e., the hovering coordinates (x, y, z, yaw) comes from the `hovering_example.cpp` file, in `rotors_gazebo\src' folder. The message is published on the `mav_msgs::default_topics::COMMAND_TRAJECTORY` topic ([here you can find the full name of the topics](https://github.com/gsilano/mav_comm/blob/crazys/mav_msgs/include/mav_msgs/default_topics.h)) to which the `PositionControllerNode` is subscribed. Whenever a message is published on this topic (for the hovering example only once) the `PositionControllerNode::MultiDofJointTrajectoryCallback` method is called. This essentially moves the information from the node to the controller algorithm which, in order to facilitate the reuse of the code, has been implemented as a shared library.

The boolean flag in the controller algorithm enables the controller. I mean, the propellers' angular velocities are equals to zero until the boolean flag rises.

https://github.com/gsilano/CrazyS/blob/59f227f4caddafbca0c2872427b7a5539f1c95a3/rotors_control/src/library/position_controller.cpp#L255-L309

Ever time a new Odometry message arrives (it contains the drone state, i.e., position, attitude, the linear and angular velocity of the drone) a new set of the propellers angular velocity is computed by the controller algorithm (`position_controlller.cpp`) in a loop and it is sent to the node (`PositionControllerNode') which forward the message to Gazebo.

I hope this helps.

KomaGR, 2020-04-12:
Thanks for your quick reply! I'm sorry if I did not explain it clearly. What you just said clarified a lot of things for me. I also looked at #10 for more information.

I think my problem is that I can't find anywhere messages being published on the Odometry topic (either on `odometry_pub_topic_` or on `mav_msgs::default_topics::ODOMETRY`) inside the source code . 

Do you know if this is something that is being handled internally in Gazebo?

> ...a new set of the propellers angular velocity is computed by the controller algorithm...

For me `PositionController::CalculateRotorVelocities` is called only when I set `enable_state_estimator` to `true`, and it is called from `PositionControllerNode::CallbackHightLevelControl`, still without any Odometry messages. Nevertheless, there is still no movement.

gsilano, 2020-04-12:
It could be, it could be. You have to consider that the `gazebo_odometry_plugin.cpp` file in `rotors_gazebo_plugins\src` manages the interface between Gazebo and the Odometry message. Thus, if the code is unable to interface with Gazebo, no Odometry messages will be published on the topic.

KomaGR, 2020-04-12:
I enabled `verbosity` in the example launch file and this is what I find:
```
[Err] [Plugin.hh:187] Failed to load plugin 
/opt/ros/melodic/share/gazebo_plugins/../../lib/librotors_gazebo_ros_interface_plugin.so: 
libmav_msgs.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:187] Failed to load plugin 
/opt/ros/melodic/share/gazebo_plugins/../../lib/librotors_gazebo_multirotor_base_plugin.so: 
libmav_msgs.so: cannot open shared object file: No such file or directory
...
```
...and so on for all `librotors_gazebo*` shared objects.

This could be very well be the source of the problem. It looks like it is similar to https://github.com/ethz-asl/rotors_simulator/pull/506 where you mention in https://github.com/ethz-asl/rotors_simulator/pull/506#issuecomment-566637382 that the drone doesn't take off. Even though your `CMakeLists.txt` contains the fix in https://github.com/ethz-asl/rotors_simulator/pull/506/commits/787f8b512f4243d5e32d800b9d9ef279ccd665f7, it doesn't seem to work correctly.

Copying the `libmav_msgs` library in the directory of the `librotors_gazebo*.so`  fixed it temporarily.
```
cp build/rotors_gazebo_plugins/libmav_msgs.so devel/lib/
```
**Of course, this is not a good fix.** The shared objects should ideally know to look for the library where it's located. Using `ldd` shows that the shared objects actually do know where it is. I do not know why the can't find it on runtime.
```
ldd catkin_wc/devel/lib/librotors_gazebo_motor_model.so | grep libmav
libmav_msgs.so => /home/koma/catkin_wc/build/rotors_gazebo_plugins/libmav_msgs.so 
(0x00007f633c6d5000)
```
-----
For any other Arch users out there, the next problem I faced was with opencv versioning. A common situation, usually resolved by linking the previous version to the newest:
```
[Err] [Plugin.hh:187] Failed to load plugin 
/opt/ros/melodic/share/gazebo_plugins/../../lib/librotors_gazebo_odometry_plugin.so: 
libopencv_imgproc.so.4.2: cannot open shared object file: No such file or directory
...
```
```
ln -s /usr/lib/libopencv_imgproc.so.4.3  /usr/lib/libopencv_imgproc.so.4.2
```
### And there was flight
[image]

gsilano, 2020-04-12:
Ok, awesome! I am happy to know that you have finally fixed the issue. So, you can run CrazyS with Gazebo 10 on an Arch Linux distro. Am I right? 

I'm thinking it would be great to put these instructions on the CrazyS and BebopS Wiki (also use RotorS as the basis for the package).

KomaGR, 2020-04-12:
> Ok, awesome! I am happy to know that you have finally fixed the issue. So, you can run CrazyS with Gazebo 10 on an Arch Linux distro. Am I right?

Yes. Current Arch Linux kernel version is `5.6.3-arch1-1` and Gazebo version is `10.1.0`.

> I'm thinking it would be great to put these instructions on the CrazyS and BebopS Wiki (also use RotorS as the basis for the package).

I'm still checking to see if there are any problems left. That said, wouldn't it be best if this was addressed at the `CMakeLists.txt`? Though, I'm not sure yet why the lib is not linking correctly.

gsilano, 2020-04-13:
Yes, of course. It would be better if we could find a way to include the library in the `CMakeList.txt` file.

AnandaNN, 2020-04-17:
This is just a note related to this topic.

I had the same problem as @KomaGR but on `Ubuntu 18.04` and using `Gazebo 9.13.0`. After reading this thread I tried doing a manual copy of libmav_msgs.so and now it can do the hover.

This problem only occurs for me when building using `catkin build` and not if it is build with `catkin_make` but I don't know what difference this makes.

gsilano, 2020-04-17:
Hi @AnandaNN! I think you can find useful issue #25 

As for the difference between `catkin build` and `catkin_make`, you can start reading [this link](https://robotics.stackexchange.com/questions/16604/ros-catkin-make-vs-catkin-build)

gsilano, 2020-04-25:
Hi @KomaGR !! I just opened a [Wiki page](https://github.com/gsilano/CrazyS/wiki/CrazyS-on-Arch-Linux-with-ROS-Melodic-and-Gazebo-10) with the instructions for installing CrazyS with Gazebo 10 and Arch Linux. Although the procedure is not optimized, I think it would be useful for others who use CrazyS with this operating system. Could you help with this?

I can't give you direct access to the Wiki page because I'm not part of an organization. GitHub doesn't allow managing collaborators if the repository is on a private account. I should create an organization, move the content of the repository, and so on. 

Therefore, I ask you, to upload the text in Markdown by using a pull request. I will then take care of the merge. In the meantime, [here you find the file](https://github.com/gsilano/CrazyS/files/4532854/Draft-Arch.txt) that I have already written.

Thanks!

gsilano, 2020-04-27:
I close this issue. A pull request (#46) has been opened to update the Wiki pages.

closed 2020-04-27 by gsilano

acxz, 2020-05-16:
Just want to pipe in that it works on the latest Gazebo (11) with `catkin_make`.