gsilano/CrazyS #41: Possibility to disable built-in control and replace it with customized algorithm

XuboLyu, 2020-04-08:
Hi @gsilano ,

Thank you so much for this great work. It does help robotics community a lot to conduct research on such classic nano-quadcopter.

Right now we are trying to do the crazyflie simulation with crazyS. However, we'd like not to use the built-in control strategy but a reinforcement learning based method. More specifically, we want to rely on the quadcopter model implemented in crazyS to evolve the state (even it will collide to the ground), and collect such data to gradually learn a control, not automatically control it using built-in PID.

So I'm wondering if it's possibile or flexible in your code to disable such built-in control, maybe reference generator as well, to allow the operations mentioned above? I'd appreciate it so much if you can give any suggestions.

Thank you! Look forward to your reply.

gsilano, 2020-04-08:
Hi @XuboLyu!! Have you read the reference papers listed in the Wiki? I think you may find some useful information on CrazyS's software architecture. However, it is possible to disable the built-in controller by commenting on the `crazyflie2_hovering_example.launch` file from line 41 to 49.

XuboLyu, 2020-04-08:
Hi @gsilano ,

Thank you so much for such quick reply. And yes, it seems directly commentig the rotor controller node in launch file makes it. :)

But I still has another question and I look into a few of crazyflie examples and do no find the answer.  Maybe I open another issue.

Thanks!

closed 2020-04-08 by XuboLyu