gsilano/CrazyS #45: Ubuntu 16.04 + ROS Kinetic + Gazebo 9 Fails Hovering Example

Fjbarchesky, 2020-04-19:
After having Gazebo7 work I decided to update to Gazebo9, however receive an error when running the following command: `roslaunch rotors_gazebo crazyflie2_hovering_example.launch`

Terminal Output: 
`[INFO] [1587323757.901873, 0.000000]: Loading model XML from ros parameter
[INFO] [1587323757.903844, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[FATAL] [1587323758.130571246]: You must call ros::init() before creating the first NodeHandle
Couldn't find an AF_INET address for []
Couldn't find an AF_INET address for []
[ERROR] [1587323758.131875188]: [registerPublisher] Failed to contact master at [:0].  Retrying...`

gsilano, 2020-04-19:
Please, try to run

```console
$ sudo apt-get remove ros-kinetic-gazebo* gazebo*
$ sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
$ wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
$ sudo apt-get update
$ sudo apt-get install gazebo9 gazebo9-* ros-kinetic-gazebo9-*
$ sudo apt upgrade
```

and then

```console
$ sudo apt-get install libeigen3-dev ros-kinetic-image-view ros-kinetic-parrot-arsdk libprotobuf-dev libprotoc-dev ros-kinetic-joy-teleop ros-kinetic-nav-msgs ros-kinetic-mav-msgs libyaml-cpp-dev ros-kinetic-nodelet ros-kinetic-mav-planning-msgs ros-kinetic-urdf ros-kinetic-image-transport ros-kinetic-roslint ros-kinetic-angles ros-kinetic-cv-bridge ros-kinetic-tf2-geometry-msgs ros-kinetic-xacro ffmpeg libavcodec-dev libavformat-dev libavutil-dev libswscale-dev ros-kinetic-camera-info-manager ros-kinetic-cmake-modules ros-kinetic-gazebo-msgs ros-kinetic-mavros-msgs ros-kinetic-control-toolbox ros-kinetic-mav-msgs ros-kinetic-libmavconn ros-kinetic-mavros ros-kinetic-octomap-msgs ros-kinetic-geographic-msgs ros-kinetic-mavlink ros-kinetic-mavros-extras ros-kinetic-mav-planning-msgs ros-kinetic-joy
```

finally

```console
$ cd ~
$ mkdir -p ros-kinetic-gazebo9-pkgs
$ cd ros-kinetic-gazebo9-pkgs
$ git clone -b feature/ros-kinetic-gazebo9-pkgs https://github.com/gsilano/BebopS.git
$ cd BebopS
$ chmod 777 gazebo9.sh
$ ./gazebo9.sh
$ cd ~
$ sudo rm -rf ros-kinetic-gazebo9-pkgs # delete the folder after the installation
$ cd ~/catkin_ws
$ catkin clean # digit y when required
$ cd ~/catkin_ws/src/CrazyS
$ git checkout dev/gazebo9
$ cd ~/catkin_ws/src/mav_comm
$ git checkout med18_gazebo9
$ cd ~/catkin_ws
$ catkin build
$ source ~/.bashrc
```

Let me know if this solve the issue.

Fjbarchesky, 2020-04-19:
I have completed repeated the process above, unfortunately I have the same result as previously mentioned.

gsilano, 2020-04-19:
Could you copy and paste the whole verbose? This can help understand what went wrong.

Fjbarchesky, 2020-04-19:
`roslaunch rotors_gazebo crazyflie2_hovering_example.launch
... logging to /home/fjb427/.ros/log/8271ae2e-8275-11ea-a56c-24ee9ae439dc/roslaunch-fjb427-asuslaptop-20866.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://fjb427-asuslaptop:39997/

SUMMARY
========

PARAMETERS
 * /crazyflie2/position_controller_node/attitude_gain_ki/phi: 0.0349
 * /crazyflie2/position_controller_node/attitude_gain_ki/theta: 0.0349
 * /crazyflie2/position_controller_node/attitude_gain_kp/phi: 0.0611
 * /crazyflie2/position_controller_node/attitude_gain_kp/theta: 0.0611
 * /crazyflie2/position_controller_node/csvFilesStoring: False
 * /crazyflie2/position_controller_node/csvFilesStoringTime: 15.0
 * /crazyflie2/position_controller_node/enable_state_estimator: False
 * /crazyflie2/position_controller_node/hovering_gain_kd/z: 373
 * /crazyflie2/position_controller_node/hovering_gain_ki/z: 3.15
 * /crazyflie2/position_controller_node/hovering_gain_kp/z: 70
 * /crazyflie2/position_controller_node/inertia/xx: 1.657171e-05
 * /crazyflie2/position_controller_node/inertia/xy: 0.0
 * /crazyflie2/position_controller_node/inertia/xz: 0.0
 * /crazyflie2/position_controller_node/inertia/yy: 1.657171e-05
 * /crazyflie2/position_controller_node/inertia/yz: 0.0
 * /crazyflie2/position_controller_node/inertia/zz: 2.9261652e-05
 * /crazyflie2/position_controller_node/mass: 0.027
 * /crazyflie2/position_controller_node/rate_gain_ki/p: 0
 * /crazyflie2/position_controller_node/rate_gain_ki/q: 0
 * /crazyflie2/position_controller_node/rate_gain_ki/r: 95.6839
 * /crazyflie2/position_controller_node/rate_gain_kp/p: 1000
 * /crazyflie2/position_controller_node/rate_gain_kp/q: 1000
 * /crazyflie2/position_controller_node/rate_gain_kp/r: 1000
 * /crazyflie2/position_controller_node/rotor_configuration/0/angle: -3.14159265359
 * /crazyflie2/position_controller_node/rotor_configuration/0/arm_length: 0.046
 * /crazyflie2/position_controller_node/rotor_configuration/0/direction: -1.0
 * /crazyflie2/position_controller_node/rotor_configuration/0/rotor_force_constant: 1.28192e-08
 * /crazyflie2/position_controller_node/rotor_configuration/0/rotor_moment_constant: 0.005964552
 * /crazyflie2/position_controller_node/rotor_configuration/1/angle: 1.57079632679
 * /crazyflie2/position_controller_node/rotor_configuration/1/arm_length: 0.046
 * /crazyflie2/position_controller_node/rotor_configuration/1/direction: 1.0
 * /crazyflie2/position_controller_node/rotor_configuration/1/rotor_force_constant: 1.28192e-08
 * /crazyflie2/position_controller_node/rotor_configuration/1/rotor_moment_constant: 0.005964552
 * /crazyflie2/position_controller_node/rotor_configuration/2/angle: 3.14159265359
 * /crazyflie2/position_controller_node/rotor_configuration/2/arm_length: 0.046
 * /crazyflie2/position_controller_node/rotor_configuration/2/direction: -1.0
 * /crazyflie2/position_controller_node/rotor_configuration/2/rotor_force_constant: 1.28192e-08
 * /crazyflie2/position_controller_node/rotor_configuration/2/rotor_moment_constant: 0.005964552
 * /crazyflie2/position_controller_node/rotor_configuration/3/angle: -1.57079632679
 * /crazyflie2/position_controller_node/rotor_configuration/3/arm_length: 0.046
 * /crazyflie2/position_controller_node/rotor_configuration/3/direction: 1.0
 * /crazyflie2/position_controller_node/rotor_configuration/3/rotor_force_constant: 1.28192e-08
 * /crazyflie2/position_controller_node/rotor_configuration/3/rotor_moment_constant: 0.005964552
 * /crazyflie2/position_controller_node/user_account: giuseppe
 * /crazyflie2/position_controller_node/xy_gain_ki/x: 5.72958
 * /crazyflie2/position_controller_node/xy_gain_ki/y: -5.72958
 * /crazyflie2/position_controller_node/xy_gain_kp/x: 3.594
 * /crazyflie2/position_controller_node/xy_gain_kp/y: -3.594
 * /crazyflie2/position_controller_node/yaw_gain_ki/yaw: 0
 * /crazyflie2/position_controller_node/yaw_gain_kp/yaw: 0.0914
 * /crazyflie2/robot_description: <?xml version="1....
 * /crazyflie2/tf_prefix: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True

NODES
  /
[... 57 lines cut ...]
Couldn't find an AF_INET address for []
Couldn't find an AF_INET address for []
[ERROR] [1587325181.342455774]: [registerPublisher] Failed to contact master at [:0].  Retrying...
Couldn't find an AF_INET address for []
Couldn't find an AF_INET address for []
Couldn't find an AF_INET address for []
Couldn't find an AF_INET address for []
Couldn't find an AF_INET address for []
Couldn't find an AF_INET address for []
Couldn't find an AF_INET address for []
Couldn't find an AF_INET address for []
Couldn't find an AF_INET address for []
Couldn't find an AF_INET address for []
[ INFO] [1587325181.883304848]: Wait for 1 second before trying to unpause Gazebo again.
Couldn't find an AF_INET address for []
Couldn't find an AF_INET address for []
[Wrn] [GuiIface.cc:119] void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&File" under id 56
[Wrn] [GuiIface.cc:119] void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&Edit" under id 59
[Wrn] [GuiIface.cc:119] void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&Camera" under id 61
[Wrn] [GuiIface.cc:119] void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&View" under id 64
[Wrn] [GuiIface.cc:119] void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&Window" under id 68
[Wrn] [GuiIface.cc:119] void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&Help" under id 73
Couldn't find an AF_INET address for []
Couldn't find an AF_INET address for []
`

gsilano, 2020-04-19:
There is something wrong with this

```
$ cd ~
$ mkdir -p ros-kinetic-gazebo9-pkgs
$ cd ros-kinetic-gazebo9-pkgs
$ git clone -b feature/ros-kinetic-gazebo9-pkgs https://github.com/gsilano/BebopS.git
$ cd BebopS
$ chmod 777 gazebo9.sh
$ ./gazebo9.sh
$ cd ~
$ sudo rm -rf ros-kinetic-gazebo9-pkgs # delete the folder after the installation
```

I mean, the procedure fails somehow. I suggest you explore the `gazebo9.sh` file and be sure that each command is executed correctly.

Fjbarchesky, 2020-04-19:
Ok, after running the gazebo9.sh script here is the output: 

`cp: omitting directory './ros/kinetic/include/gazebo_msgs'
cp: omitting directory './ros/kinetic/include/gazebo_plugins'
cp: omitting directory './ros/kinetic/include/gazebo_ros'
cp: omitting directory './ros/kinetic/lib/gazebo_plugins'
cp: omitting directory './ros/kinetic/lib/gazebo_ros'
cp: omitting directory './ros/kinetic/share/gazebo_dev'
cp: omitting directory './ros/kinetic/share/gazebo_msgs'
cp: omitting directory './ros/kinetic/share/gazebo_plugins'
cp: omitting directory './ros/kinetic/share/gazebo_ros'
cp: omitting directory './ros/kinetic/share/gazebo_ros_control'
cp: omitting directory './ros/kinetic/share/gazebo_ros_pkgs'
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Add distro "dashing"
Add distro "eloquent"
Add distro "foxy"
Skip end-of-life distro "groovy"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Add distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
updated cache in /home/fjb427/.ros/rosdep/sources.cache
fjb427@fjb427-asuslaptop:~/ros-kinetic-gazebo9-pkgs/BebopS$ 
`

Are the "omitting"'s normal?

gsilano, 2020-04-19:
Yes, omitting is normal. Please, continue with the steps reported in the guide. I think the problem is related to the installation of Gazebo or ROS. Have you installed the full version of ROS?
As you can see from the verbose, it is looking for the `libgazebo_ros_paths_plugin.so` object file. Did you source later the installation?

```
[Err] [Plugin.hh:178] Failed to load plugin /opt/ros/kinetic/lib/libgazebo_ros_paths_plugin.so: libgazebo_common.so.7: cannot open shared object file: No such file or directory
```

Fjbarchesky, 2020-04-19:
I have removed and re-installed ROS and Gazebo (multiple times). Definitely installed the full version of ROS per your instructions. 

What you mean by "did you source later the installation"?

gsilano, 2020-04-19:
```
$ echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
$ source ~/.bashrc
```

As you can see from https://github.com/gsilano/CrazyS/issues/24#issuecomment-574728684, others had the same problem.

Fjbarchesky, 2020-04-19:
I have executed those commands per your instructions and my './bashrc' file seems to be correct.

gsilano, 2020-04-19:
In order to understand what goes wrong, I suggest to uninstall both ROS and Gazebo, delete them folders in the OS, and try again with Gazebo 7. If it works, we will use it as a starting point for Gazebo 9.

Fjbarchesky, 2020-04-19:
I will uninstall both. 

Previously I had a Gazebo7 running correctly and was able to execute the simple hovering example, without any errors. It was only after the Gazebo9 changes where this proceeded to fault.

Fjbarchesky, 2020-04-19:
Ran the following commands: 

`sudo apt-get purge '^ros-*'
sudo apt-get remove --auto-remove gazebo*
sudo apt-get purge --auto-remove gazebo*
sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
sudo apt-get update
sudo apt-get install ros-kinetic-desktop-full ros-kinetic-joy ros-kinetic-octomap-ros ros-kinetic-mavlink python-catkin-tools protobuf-compiler libgoogle-glog-dev ros-kinetic-control-toolbox
sudo rm /etc/ros/rosdep/sources.list.d/20-default.list 
sudo rosdep init
echo "source /opt/ros/kinetic/setup.bash" >> ~/.bashrc
rosdep updatesource ~/.bashrc
sudo apt-get install python-rosinstall python-rosinstall-generator python-wstool build-essential
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src
catkin_init_workspace 
cd ~/catkin_ws/
catkin init
cd ~/catkin_ws/src
git clone https://github.com/gsilano/CrazyS.git
git clone -b crazys https://github.com/gsilano/mav_comm.git
cd ~/catkin_ws
rosdep install --from-paths src -i
catkin build
echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
source ~/.bashrc

@ this point I ran the hovering example and Gazebo showed the crazyflie in a hover. 

sudo apt-get remove ros-kinetic-gazebo* gazebo*
sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
sudo apt-get update
sudo apt-get install gazebo9 gazebo9-* ros-kinetic-gazebo9-*
sudo apt upgrade
sudo apt-get install libeigen3-dev ros-kinetic-image-view ros-kinetic-parrot-arsdk libprotobuf-dev libprotoc-dev ros-kinetic-joy-teleop ros-kinetic-nav-msgs ros-kinetic-mav-msgs libyaml-cpp-dev ros-kinetic-nodelet ros-kinetic-mav-planning-msgs ros-kinetic-urdf ros-kinetic-image-transport ros-kinetic-roslint ros-kinetic-angles ros-kinetic-cv-bridge ros-kinetic-tf2-geometry-msgs ros-kinetic-xacro ffmpeg libavcodec-dev libavformat-dev libavutil-dev libswscale-dev ros-kinetic-camera-info-manager ros-kinetic-cmake-modules ros-kinetic-gazebo-msgs ros-kinetic-mavros-msgs ros-kinetic-control-toolbox ros-kinetic-mav-msgs ros-kinetic-libmavconn ros-kinetic-mavros ros-kinetic-octomap-msgs ros-kinetic-geographic-msgs ros-kinetic-mavlink ros-kinetic-mavros-extras ros-kinetic-mav-planning-msgs ros-kinetic-joy
cd ~
mkdir -p ros-kinetic-gazebo9-pkgs
cd ros-kinetic-gazebo9-pkgs
git clone -b feature/ros-kinetic-gazebo9-pkgs https://github.com/gsilano/BebopS.git
cd BebopS
chmod 777 gazebo9.sh
./gazebo9.sh
cd ~
sudo rm -rf ros-kinetic-gazebo9-pkgs # delete the folder after the installation
cd ~/catkin_ws
catkin clean # digit y when required
cd ~/catkin_ws/src/CrazyS
git checkout dev/gazebo9
cd ~/catkin_ws/src/mav_comm
git checkout med18_gazebo9
cd ~/catkin_ws
catkin build
source ~/.bashrc`

Ran the hover example again, and received the identical fault/error as before.

gsilano, 2020-04-20:
Try with this

```
cd ~/catkin_ws/src/
rosdep check --from-paths src into the catkin_ws
```

and then ran again the `hovering_example` launch file.

Fjbarchesky, 2020-04-20:
I was unable to run the specific commands you dictated due to an error. I was able to run: 

`cd ~/catkin_ws/
rosdep check --from-paths src`

Which I was then able to get the following output: 

System dependencies have not been satisfied:
apt	gazebo7
apt	ros-kinetic-gazebo-plugins
apt	ros-kinetic-gazebo-ros
apt	ros-kinetic-gazebo-dev
ERROR[rotors_simulator]: Cannot locate rosdep definition for [rqt_rotors]
	rosdep key : rqt_rotors
	OS name    : ubuntu
	OS version : xenial
	Data:
<no data>
ERROR[rotors_gazebo_plugins]: Cannot locate rosdep definition for [rotors_control]
	rosdep key : rotors_control
	OS name    : ubuntu
	OS version : xenial
	Data:
<no data>
ERROR[rotors_gazebo]: Cannot locate rosdep definition for [rotors_gazebo_plugins]
	rosdep key : rotors_gazebo_plugins
	OS name    : ubuntu
	OS version : xenial
	Data:
<no data>
ERROR[mav_comm]: Cannot locate rosdep definition for [mav_system_msgs]
	rosdep key : mav_system_msgs
	OS name    : ubuntu
	OS version : xenial
	Data:
<no data>

gsilano, 2020-04-20:
Of the listed dependencies, only one is related to Gazebo 7. Of course, since you want to use Gazebo 9 on ROS Kinetic, you don't need to install Gazebo 7. Instead, you need to install the following packages

```console
$ sudo apt-get install ros-kinetic-gazebo9-plugins
$ sudo apt-get install apt ros-kinetic-gazebo9-ros
$ sudo apt-get install apt ros-kinetic-gazebo9-dev
```

Fjbarchesky, 2020-04-20:
We have successful execution of the hovering example now! thank you for your assistance.

closed 2020-04-20 by Fjbarchesky