gsilano/CrazyS #47: Unable to move in XY direction with DroneState message

senceryazici, 2020-05-01:
Hi

with the dronestate message I'm trying to achieve a point, but since the orientation reference (+ angular_vel + angular_acc) are zero, and only position X-Y (or +linear_vel) is changing, I'm unable to move the vehicle. The linear z movement is working properly. 

How can I move the CrazyFlie to an XYZ point, with DroneState message ?

gsilano, 2020-05-01:
The `DroneState` message contains the desired reference trajectory. I know that the message name is misleading and perhaps I should change it but, at first, it was not thought to work with the Internal Model Controller. 

However, as you can see from the code in the launch file, i.e., the `crazyflie2_internal_model_controller.launch`, `rotors_gazebo` folder, a spline generator function is used to generate the desired references. Therefore, to generate a reference trajectory other than the hovering example you need to change the parameters in the yaml file, i.e., `spline_trajectory.yaml`.

Please, keep in mind that the controller gains could be tuned to perform the trajectory tracking problem.

senceryazici, 2020-05-02:
Thanks @gsilano , I'll try that.

Bot is cute btw :)

closed 2020-05-02 by senceryazici

sunnyshi0310, 2021-02-01:
Hi, I want to follow that problem.
So are there any way that I could publish several desired waypoints to the robot? It seems that I could only assign one waypoint in the `spline_trajectory.yaml` file and can not move it to other position once I launch the file. I expect to launch it to (0,0,1) and then publish the trajectory to topic `drone_state` using `DroneState` msg, while it doesn't work. I appreciate any help, thanks.

gsilano, 2021-02-01:
Hi @sunnyshi0310! The integrated controllers, i.e., the `internal_model.cpp` and `position_controller.cpp` are not meant as trajectory tracking controllers. The main motivation behind the repository was to create a valuable simulator of the Crazyflie 2.0 nano quad-rotor. Thus, the focus was on modeling the aircraft behavior and its onboard sensors.

As for your question, I took a look at the scripts in the `rotors_gazebo` folder. The `waypoint_publisher_file.cpp` file should be the script you are looking for. The script allows publishing the waypoints listed in the `example_waypoints.txt` file (`rotors_gazebo\resource` folder).

Now, if you are interested in generating splines starting from a set of waypoints, I think is not that complicated. You just need to modify the `waypoint_publisher_file.cpp` script using `spline_trajectory_generator.cpp` as library for the trajectory generation. Roughly speaking, you just have to pass as arguments to the spline the waypoints in the txt file until there is another waypoint in the queue.

I hope I was enough clear.

sunnyshi0310, 2021-02-01:
Hi, Thanks for your explanation. So you mean we could change the point in the `example_waypoints.txt` and launch the `mav_with_waypoint_publisher.launch
` (maybe not this one, I am not sure) so it could go to the positions we specify in the `example_waypoints.txt`? While we are actually doing a task allocation project so we want to update the desired poses online instead of writing them in a predefined file. Could we add another subscriber in the `waypoint_publisher_file.cpp` or do you have other suggestions on achieving that goal? Thanks again.

gsilano, 2021-02-02:
You can put this line `https://github.com/gsilano/CrazyS/blob/master/rotors_gazebo/launch/mav_with_waypoint_publisher.launch#L31` in each launch file. You have to deal with the I/O interface, but then it should work. 

As I said, the controller is not meant as a trajectory tracking controller. So, it might not work. If you are still interested in publishing various waypoints, this is the fastest solution I can suggest to you.

You can follow the structure of the cpp file (i.e., `spline_trajectory_generator.cpp`) or see how the values of the spline are published along the time and over the network to understand how to manage it. This should be not so that hard. 

If you want to help with a specific question, I think it would be better to open a new thread. This way, your questions, and my answers may be of help also to other community members.