gsilano/CrazyS #48: Adding a camera to crazyS

ziyadAbouelenin, 2020-05-01:
Hello @gsilano , 

I want to add a camera to the crazyFlie, any idea how to do this? 

I found the [Adding a Camera to CrazyS] part in the wiki, but can you explain more or just point me to a reference if any. 

Thank you

gsilano, 2020-05-02:
Hi! Thanks for your interest in CrazyS. As for your question, as it was complicated to explain how to add a simulated camera with CrazyS, I thought it was better to develop a sample example scenario.

As you can see from the commit above, I added the `crazyflie2_internal_model_controller_vi_sensor.launch` file to simulate Crazyflie with a VI sensor. To this, I added an RVIZ configuration file to see the camera images and the point cloud data.

I hope this can help.

closed 2020-05-02 by gsilano

ziyadAbouelenin, 2020-05-02:
Thank you. This is very helpful.