gsilano/CrazyS #49: Octomap generation

Amit10311, 2020-05-04:
Thank you for this great package for simulating MAV. I am running Ubuntu 16.04 with ROS kinetic and Gazebo9. The simulator is working perfectly. I have added a plugin of octomap in my .world file but the service is not activated.  

I look forward to hearing from you. Thank you for your valuable time.

gsilano, 2020-05-06:
Hi@Amit10311! When you say that the Octomap plugin is not working, you are referring to [this guide](https://github.com/gsilano/CrazyS/wiki/Generate-an-octomap-from-your-world). Am I right?

Amit10311, 2020-05-06:
Hello, Yes I am referring  [the link ](https://github.com/gsilano/CrazyS/wiki/Generate-an-octomap-from-your-world) mentioned in the wiki .

gsilano, 2020-05-10:
Sorry for the late reply. I got the issue and updated the instructions in the Wiki. Currently, the Octomap plugin only works with Gazebo 7 and Ubuntu 16.04. I'm working to make it compatible with Gazebo 9 too.

Amit10311, 2020-05-12:
Thanks for the updates, can you please tell me estimate days to prepare plugins for gazebo9. 
And I have one more question, is it possible to attach the given lidar to the drone and make navigation.

gsilano, 2020-05-12:
What do you mean "estimated days to prepare plugins for Gazebo9"? Yes, it is possible to equip the Crazyflie with a LiDAR. You could follow the code I used to attach a VI sensor to the Crazyflie 2.0 nano-quadrotor.

Amit10311, 2020-05-12:
Thanks again for the LIDAR info. I meant before, as you're working on octomap plugins, so it will be available soon, I know it usually takes time.

gsilano, 2020-05-12:
Yes, of course. It takes time. In the meantime, I could provide you a workaround solution. I mean, Octomap and RVIZ that work together for generating a 3D map of the environment. Let me know if this is ok with you.

Amit10311, 2020-05-12:
Yes, it will be ok for me, please provide the solution.

gsilano, 2020-05-13:
I just added a simple scenario for the Crazyflie with Octomap. It is available in `crazyflie2_octomap.launch`. Before using it, run the command `sudo apt-get install ros-$ROS_DISTRO-octomap-mapping` where `$ROS_DISTRO` is your ROS distro, i.e., kinetic, melodic.

gsilano, 2020-05-24:
Hi @Amit10311 ! Did my question solve the issue? If so, I will close the issue. However, I suggest you take a look at [this repository](https://github.com/larics/larics_gazebo_worlds). It might help you in generating an Octomap scenario.

Amit10311, 2020-05-24:
Hi @gsilano ! Yes, it solved the issue. Thanks for the suggestion, it's also a good and quick solution.  Yes, you can close the issue. Thanks for the quick solutions.

closed 2020-05-24 by gsilano