gsilano/CrazyS #50: problem with crazyflie2 hovering example

Rucy-kim, 2020-05-11:
Hi!

Thank you for this package for simulation.

I am running Ubuntu 16.04 with ROS Kinetic and Gazebo7 right now.
However, when I ran the hovering example, the output seems to be incorrect.

I tested these examples:
```
roslaunch rotors_gazebo crazyflie2_hovering_example.launch
roslaunch rotors_gazebo crazyflie2_hovering_example.launch enable_state_estimator:=true
roslaunch rotors_gazebo crazyflie2_internal_model_controller.launch
```
and the only example I think is the correct result is the internal model launcher.

crazyflie2_hovering_example.launch takes off, but doesn't stop at 1m and loses control and moves randomly.

And in the case of enable_state_estimator: = true, the propeller is running slowly, but does not take off.

In all three cases, no error message was displayed.
Is there any other information I can refer to as something?

Thanks in advance for the help.

gsilano, 2020-05-11:
It seems very strange. Could you show me a video?

Rucy-kim, 2020-05-11:
sure.
This is link of my youtube.

I filmed a video with my mobile phone.
The picture quality is not good, but I think you can confirm the phenomenon.

[crazyflie2_hovering_example.launch](https://www.youtube.com/watch?v=T-V6YSuDGOE)
[crazyflie2_hovering_example.launch enable_state_estimator:=true](https://www.youtube.com/watch?v=uryxN2DIuBs)

gsilano, 2020-05-11:
Pull the last changes, and let me know. If it solves the issue, I will propagate the changes in the other releases, i.e., ROS Kinetic with Gazebo 9 and ROS Melodic.

Rucy-kim, 2020-05-11:
Thanks for the quick response.
I tried again following your suggestion, but unfortunately it didn't work.
This is the same behavior as before.

gsilano, 2020-05-11:
Sorry, I modified the `ros_melodic` branch. Try now.

Rucy-kim, 2020-05-11:
I pulled the `dev/ros-melodic` branch in the previous test and tested it, is it the same code?
> Sorry, I modified the `ros_melodic` branch. Try now.

And I tested both ros_melodic and master branch now, but unfortunately it was not fixed.

gsilano, 2020-05-11:
It seems strange because it works for me. Could you copy and paste the contents of the `basic_crazyflie.world` file? It is located in the `rotors_gazebo\worlds` folder.

Rucy-kim, 2020-05-11:
Ok, this is my `basic_crazyflie.world` file

```
<?xml version="1.0" ?>
<?xml-model href="http://sdformat.org/schemas/root.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<sdf version="1.4">
  <world name="default">
    <include>
      <uri>model://ground_plane</uri>
    </include>
    <include>
      <uri>model://sun</uri>
    </include>

    <!-- Only one ROS interface plugin is required per world, as any other plugin can connect a Gazebo
         topic to a ROS topic (or vise versa). -->
    <plugin name="ros_interface_plugin" filename="librotors_gazebo_ros_interface_plugin.so"/>

    <!-- coordinate system {-->
    <spherical_coordinates>
      <surface_model>EARTH_WGS84</surface_model>
      <latitude_deg>47.3667</latitude_deg>
      <longitude_deg>8.5500</longitude_deg>
      <elevation>500.0</elevation>
      <heading_deg>0</heading_deg>
    </spherical_coordinates>
    <!--}-->

    <!-- physics engine {-->
    <physics name="default_physics" default="0"  type='ode'>
      <ode>
        <solver>
          <type>quick</type>
          <iters>1000</iters>
          <sor>1.3</sor>
        </solver>
        <constraints>
          <cfm>0</cfm>
          <erp>0.2</erp>
          <contact_max_correcting_vel>100</contact_max_correcting_vel>
          <contact_surface_layer>0.001</contact_surface_layer>
        </constraints>
      </ode>
      <max_step_size>0.001</max_step_size>
      <real_time_factor>1</real_time_factor>
      <real_time_update_rate>1000</real_time_update_rate>
      <magnetic_field>6.0e-06 2.3e-05 -4.2e-05</magnetic_field>
      <gravity>0 0 -9.8</gravity>
    </physics>
    <!--}-->

    <!-- turn off shadows {-->
    <scene>
      <shadows>false</shadows>
      <sky>
        <clouds/>
      </sky>
    </scene>
    <!--}-->

  </world>
</sdf>
```

gsilano, 2020-05-11:
Okay, that's how it should be. Hmm. Did you run the `catkin clean` command and then `catkin build` again?

Rucy-kim, 2020-05-11:
Oh..I don't try re-build yet.
I will try it and let you know.

Rucy-kim, 2020-05-11:
and I'm so sorry,.
This is now 3 am, so the results will be announced tomorrow.

Rucy-kim, 2020-05-12:
HI, gsilano
Sorry to late reply and thank you for your quick fix.

I now want to use CrazyS to simulate [crazyswarm](https://github.com/USC-ACTLab/crazyswarm) so I tested on three cases.
 
1. install CrazyS on ros_ws directory in crazyswarm and use `crazyswarm/build.sh`
-> 
This has the same problem as I asked above.
Although this build script is configured to use catkin_make, it may be different, but I am not sure.

2. install CrazyS on ros_ws directory in crazyswarm and use `catkin build`
->  
`roslaunch rotors_gazebo crazyflie2_hovering_example.launch`

This example seems to repeat rising and falling without changing the x and y axes.
After ascending to a very high position, descending, hovering for a while, and then rising again.
Is this correct?

`roslaunch rotors_gazebo crazyflie2_hovering_example.launch enable_state_estimator:=true`

Unfortunately, this example still has problems.
It appears to be a successful take-off, but when it reaches a certain altitude, it loses control and crashes.

3. install CrazyS on catkin_ws and use `catkin build`
->
This is an error in the build, so I couldn't actually test the example.
There seems to be a conflict with crazyswarm's ros workspace, but I can't understand it.

It seems unlikely that this is not a problem for everyone.
I also plan to reinstall ros and test CrazyS.
If you know how to clean it, please help us and we will do it.

Thanks, Regards.

gsilano, 2020-05-12:
I am happy to hear from you that gsilano@1b3eb1a420466fa8704e1a2fc55d82be1cb3cdce has fixed the problem. Now, since your question is no longer related to the `hovering example`, it would be better to open a new issue by closing this. In fact, the reader who may have the same problem may be confused by reading this conversation if we mix everything up.

closed 2020-05-12 by gsilano

Rucy-kim, 2020-05-12:
Sorry, Did you read my comment?

I didn't say that the [1b3eb1a](https://github.com/gsilano/CrazyS/commit/1b3eb1a420466fa8704e1a2fc55d82be1cb3cdce)  fixed the problem.

Or is the situation written in 2. correct?

gsilano, 2020-05-12:
Sorry, perhaps I did not get it right. You wrote "thanks for your quick fix", so I thought that now the problem was now fixed.

Please try to separate the problems. First of all, let's understand if there are still problems with CrazyS. So we will discuss integrating CrazyS with the `crazyswarm` ROS package.