gsilano/CrazyS #51: hovering example incorrect

Rucy-kim, 2020-05-13:
Hi, gsilano
I apologize for confusing you with a difficult answer to understand in the # 50 issue.

I wanted to check where the problem is, so I tried the following for:

1. Delete all ros related packages
2. Delete crazyswarm related code and workspace
3. CrazyS reinstall
(I did all of ubuntu16.04 with gazebo7 installation process according to your README.md)
4. execute example code:
```
roslaunch rotors_gazebo crazyflie2_hovering_example.launch
roslaunch rotors_gazebo crazyflie2_hovering_example.launch enable_state_estimator:=true
```
Example code execution was completed both before and after [1b3eb1a](https://github.com/gsilano/CrazyS/commit/1b3eb1a420466fa8704e1a2fc55d82be1cb3cdce) .

But still the problem is not solved and the simulation seems to be inaccurate.
Is there any information I need to give you more to solve this?

gsilano, 2020-05-13:
As you can see from [this video](https://youtu.be/jay7AU2VzvI), it works for me. However, I'm working with Gazebo 9 on ROS Melodic. You could try installing Gazebo 9 instead of using Gazebo 7 (the instructions are reported in the [README.md file](https://github.com/gsilano/CrazyS/blob/master/README.md)). Perhaps, this solves the issue.

Rucy-kim, 2020-05-14:
Thank you for showing the video on purpose.
On ubuntu 16.04, ros kinetic, gazebo9 you can see that it moves similar to your video.
Thank you for your help.

closed 2020-05-14 by Rucy-kim