gsilano/CrazyS #53: How to develop on CrazyS such that the code is 1-to-1 with hardware code?

acxz, 2020-05-22:
Thanks for making this amazing simulator!

Since I am developing code on the hardware crazyflie I would like to be able write the same controller code for this simulator and the hardware. What is the recommended approach to do this?

gsilano, 2020-05-24:
Hi @acxz!! I don't know if I understand your question correctly. However, if you aim to control the nano-quadrotor using a ROS node, such as [crazyflie_ros](https://github.com/whoenig/crazyflie_ros), there is no need to change anything in your control architecture other than the unit measurements interface. I mean, if you see, in crazyflie_ros the thrust force is expressed as INT16 number, while in CrazyS the thrust is a force (Newton).

Therefore, I suggest that you start reading the [reference publications](https://github.com/gsilano/CrazyS/wiki/Publications), and then the crazyflie_ros documentation. This way, you can get a clearer idea of how the systems work, and where changes need to be applied (if there are changes to be made).

acxz, 2020-05-24:
Thanks for pointing me to crazyflie_ros and the publications! 

Is there anyway to add a control mode which replicates the unit measurement interface in this codebase? This would be nice, since I could use this simulator to verify my hardware code to a greater degree.

gsilano, 2020-05-25:
I did not understand what you mean. Anyway, reading the papers I have indicated, you will immediately understand how to build this interface for your purposes.

acxz, 2020-06-01:
Sorry for getting back so late @gsilano 
Okay after some reading and discussing with @mouhyemen, I think I figured what needs to be done.

For achieving 1-to-1 hardware code compatibility with CrazyS, CrazyS should subscribe to messages on the `/cmd_vel` topic See: https://github.com/whoenig/crazyflie_ros#cmd_vel
This will allow code that is written for crazyflie-ros (hardware code) to also be used with CrazyS (simulation code).

From what I understand right now, CrazyS only subscribes to `/crazyflie2/command/motor_speed`. Would it be possible to add the ability to subscribe to `/cmd_vel` with the topic information as linked above from crazyflie_ros from this repo as well?

gsilano, 2020-06-02:
HI @acxz!! The CrazyS interface is slightly different from the `crazyflie_ros` package provided by Whoening. I mean, the platform is designed to test, in a very high-level stage, algorithms performed in a MIL (Model-in-the-loop) scenario, e.g., MATLAB/Simulink. Hence, it aims to help students and researchers who would like to test their algorithms in an environment as close to the real world as possible.

I can perfectly understand that you would like to have a 1-to-1 simulator, using the same ROS package you will employ during the real-experiment, but so far CrazyS has not implemented such interface. It would be nice and also useful for the whole community if you could help in organizing such an environment. If you think it might be useful, we could organize a call to describe the details.

Unfortunately, this period is very stressful for me. I'm finishing my doctorate and preparing for the next stage. I hope you can understand.

acxz, 2020-06-02:
I do have a couple of thoughts on how to create this interface. Basically create some logic, which I believe you have already done in your roll-pitch-yawrate-thrust-controller, that converts roll-pitch-yawrate-thrust commands from the same message type of `/cmd_vel` that crazyflie_ros uses into CrazyS's `/motor_speeds. I can make a very naive implementation of this as a separate ROS package, but would be unsure how to merge this interface properly with CrazyS. For now I'll just create this simple interface and then we can see how this feature can be part of CrazyS directly. I propose we keep this issue open until I create this naive interface implementation. Thx for your advice. 

> I'm finishing my doctorate and preparing for the next stage

Good luck!

gsilano, 2020-08-02:
@acxz, any updates? are you working or are you going to work on this? Otherwise, I will close the issue. Of course, we can reopen it if necessary.

acxz, 2020-08-02:
Sadly I don't have updates, lets keep it closed for know, until someone expresses interest again.

closed 2020-08-02 by gsilano