gsilano/CrazyS #55: Flight interface

YuchaoS, 2020-05-30:
Hi, gsilano
I'm sorry to disturb you, but I have the following questions to solve:

1. I changed the `desired_position` from`(0,0,1)` to `(-10,-10,2)` in `hovering_example.cpp`，but only works on the z axis.
2.I want to use this project to do drone navigation, how can I perform continuous position control or speed control?

gsilano, 2020-05-31:
Hi @YuchaoS!! I suggest reading #43, and then come back to me if you have any more questions. I hope this helps.

YuchaoS, 2020-05-31:
Thank you for your reply. 
I read this, but I have no knowledge of automation control. 
I just want to use this as an algorithm verification platform (through some advanced interfaces), do you have any suggestions?

gsilano, 2020-05-31:
I do not understand how you would like to use CrazyS. However, the actual position controller is unable to perform trajectory tracking. I mean, if you change the desired position from `(x, y, z) = (0, 0, 1)` to `(x, y, z) = (-10, 10, 5)`, it will not reach never this point. In fact, as it is written in the [reference papers](https://github.com/gsilano/CrazyS/wiki/Publications), the platform is meant to be a software-in-the-loop tool aimed to simulate the behavior of the Crazyflie 2.0 nano-quadrotor. Researchers and developers can use it as is or can improve its functionalities by adding, for example, more sophisticated control algorithms.

Therefore, if you could be more precise, we could find a way to address your issue.

YuchaoS, 2020-05-31:
Thank you for your reply. 
My idea is that the drone can fly around to the target coordinate point by sensing the surrounding obstacles, and then the input to the drone can be continuous or discrete speed control，or waypoints. Does CrazyS provide  APIs in this regard or, what should I do on this basis?

gsilano, 2020-05-31:
Hmm. Unfortunately, CrazyS does not currently implement any controller capable for tracking a trajectory or a sequence of waypoints.

closed 2020-08-02 by gsilano