gsilano/CrazyS #56: Parameters in crazyflie2.xacro

Tonirayn, 2020-07-19:
Looking at the parameters in the file crazyflie2.xacro I noticed that the drag coefficient is the same of the Iris drone which is huger that the crazyflie2. is the parameter correct? In addition also the other parameters are significantly different from those in the Forster

gsilano, 2020-07-20:
The parameter values of the Crazyflie have been taken from [this repository](https://github.com/USC-ACTLab/rotors_simulator/tree/crazyflie-dev) by the same research group of

```
W. Honig and N. Ayanian, “Flying Multiple UAVs using ROS,” in ¨ Robot Operating System (ROS): The Complete Reference (Volume 2), A. Koubaa, Ed. Springer International Publishing, 2017, pp. 83–118.
```

For more details, see the [reference book chapter](https://github.com/gsilano/CrazyS/wiki/Publications).

Tonirayn, 2020-07-20:
Yeah I have already checked the papers but I found strange that the [drag](https://github.com/gsilano/CrazyS/blob/8b9689c681195b3abb2a31e3850046eeeadcfc2d/rotors_description/urdf/crazyflie2.xacro#L43) is so high since it is equal to the [IRIS]( https://github.com/gsilano/CrazyS/blob/8b9689c681195b3abb2a31e3850046eeeadcfc2d/rotors_description/urdf/iris.xacro#L27)

gsilano, 2020-07-20:
Yes, you are right. Probably such a value is not fair but, first, it would be of interest to understand how much this value can influence the system dynamics (take a look at the [Gazebo plugins](https://github.com/gsilano/CrazyS/tree/master/rotors_gazebo_plugins)) and then if someone already addressed its estimation. If you could investigate it, it would be awesome.

Tonirayn, 2020-07-20:
Ok I have already done some experiments for making flip where the drag take much more influence and I think that a proper value it will be  1.066428e-06 according to Forster.  Indeed it is used [here](https://github.com/gsilano/CrazyS/blob/8b9689c681195b3abb2a31e3850046eeeadcfc2d/rotors_gazebo_plugins/src/gazebo_motor_model.cpp#L296)

gsilano, 2020-07-20:
Okay, thanks! I will test it as soon as I have time and let you know if this somehow affects the existing control algorithms. So, I will integrate the new value into the repository.

Tonirayn, 2020-07-20:
Just to be clear maybe It's worth also to update the other values,also according to the Forster, as follow: 

 
  <xacro:property name="motor_constant" value="1.71465181e-08" /> 
  <xacro:property name="moment_constant" value="0.004459273" /> 
  <xacro:property name="time_constant_up" value="0.0125" /> 
  <xacro:property name="time_constant_down" value="0.025" /> 
  <xacro:property name="max_rot_velocity" value="3052" /> 
  <xacro:property name="rotor_drag_coefficient" value="1.066428e-06" />
  <xacro:property name="rolling_moment_coefficient" value="0" />

gsilano, 2020-07-20:
Just to simplify the check process, could you also indicate the pages where you found these values?

Tonirayn, 2020-07-21:
sure, in the overview result page 57

gsilano, 2020-08-15:
Hi @Tonirayn!! Sorry if I did not reply to you, but this very is very full of work for me. I will manage to test your suggestions as soon as possible.

Tonirayn, 2020-08-25:
ok thanks