gsilano/CrazyS #57: Not loading world file from RotorS package

aka-LJ, 2020-08-06:
Hello,

I am on Melodic + 18.04.

I try to change the world file for the MAV hovering example, but I can only load the basic world and Yosemite one.

could you please suggest to me what adaption do I need to load the outdoor world.

aka-LJ, 2020-08-06:
Here is my world file.

```
<?xml version="1.0" ?>
<sdf version="1.4">
  <world name="outdoor">
    <!-- Place user camera -->
    <gui>
      <camera name="user_camera">
        <pose>-11.660200 -5.942800 6.053900 0.000000 0.443643 0.352194</pose>
      </camera>
    </gui>
    <scene>
      <ambient>0.2 0.2 0.2 1.0</ambient>
      <grid>false</grid>
      <sky>
        <clouds>
          <speed>4</speed>
        </clouds>
      </sky>
    </scene>

    <!-- Only one ROS interface plugin is required per world, as any other plugin can connect a Gazebo
         topic to a ROS topic (or vise versa). -->
    <plugin name="ros_interface_plugin" filename="librotors_gazebo_ros_interface_plugin.so"/>
    
    <spherical_coordinates>
      <surface_model>EARTH_WGS84</surface_model>
      <latitude_deg>47.3667</latitude_deg>
      <longitude_deg>8.5500</longitude_deg>
      <elevation>500.0</elevation>
      <heading_deg>0</heading_deg>
    </spherical_coordinates>
    <physics type='ode'>
      <ode>
        <solver>
          <type>quick</type>
          <iters>1000</iters>
          <sor>1.3</sor>
        </solver>
        <constraints>
          <cfm>0</cfm>
          <erp>0.2</erp>
          <contact_max_correcting_vel>100</contact_max_correcting_vel>
          <contact_surface_layer>0.001</contact_surface_layer>
        </constraints>
      </ode>
      <max_step_size>0.01</max_step_size>
      <real_time_factor>1</real_time_factor>
      <real_time_update_rate>100</real_time_update_rate>
      <gravity>0 0 -9.8</gravity>
    </physics>
    <!-- Light Source -->
    <light type="directional" name="sun">
      <cast_shadows>false</cast_shadows>
      <pose>0 0 100 0 0 0</pose>
      <diffuse>0.8 0.8 0.8 1</diffuse>
      <specular>0.9 0.9 0.9 1</specular>
      <attenuation>
        <range>1000</range>
        <constant>0.9</constant>
        <linear>0.01</linear>
        <quadratic>0.001</quadratic>
      </attenuation>
      <direction>0.2 0.2 -0.9</direction>
    </light>
    <light type="point" name="light_bulb">
      <cast_shadows>false</cast_shadows>
      <pose>-14 0 5 0 0 0</pose>
      <diffuse>0.3 0.3 0.3 1</diffuse>
[... 289 lines cut ...]
              </ode>
            </friction>
          </surface>
        </collision>
        <collision name="collision_5">
          <pose>198 106 0 0 0 0</pose>
          <geometry>
            <box>
              <size>6 5 0.02</size>
            </box>
          </geometry>
          <surface>
            <friction>
              <ode>
                <mu>0.8</mu>
                <mu2>0.8</mu2>
              </ode>
            </friction>
          </surface>
        </collision>
      </link>
    </model>
  </world>
</sdf>
```

gsilano, 2020-08-07:
Ok. Let's suppose your world file makes sense. What I'm wondering how did you modify the launch file to work with it? Could you copy and past it?

aka-LJ, 2020-08-11:
Sorry for the late reply, here is the launch file

`
<launch>
  <arg name="mav_name" default="firefly"/>
  <arg name="world_name" default="test_city_2"/>
  <arg name="enable_logging" default="false" />
  <arg name="enable_ground_truth" default="true" />
  <arg name="log_file" default="$(arg mav_name)" />
  <arg name="path" default=""/>
  <arg name="changeIdFrame" default="/talker/changes"/>

  <env name="GAZEBO_MODEL_PATH" value="${GAZEBO_MODEL_PATH}:$(find rotors_gazebo)/models"/>
  <env name="GAZEBO_RESOURCE_PATH" value="${GAZEBO_RESOURCE_PATH}:$(find rotors_gazebo)/models"/>
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" value="$(find rotors_gazebo)/worlds/$(arg world_name).world" />
    
    <arg name="paused" value="true"/>
    
  </include>

  <group ns="$(arg mav_name)">
    <include file="$(find rotors_gazebo)/launch/spawn_mav.launch">
      <arg name="mav_name" value="$(arg mav_name)" />
      
      <arg name="model" value="$(find rotors_description)/urdf/mav_with_vi_sensor.gazebo" />

      <arg name="enable_logging" value="$(arg enable_logging)" />
      <arg name="enable_ground_truth" value="$(arg enable_ground_truth)" />
      <arg name="log_file" value="$(arg log_file)"/>

    </include>
    <node name="lee_position_controller_node" pkg="rotors_control" type="lee_position_controller_node" output="screen">
      <rosparam command="load" file="$(find rotors_gazebo)/resource/lee_controller_$(arg mav_name).yaml" />
      <rosparam command="load" file="$(find rotors_gazebo)/resource/$(arg mav_name).yaml" />
      <remap from="odometry" to="odometry_sensor1/odometry" />
    </node>

    <node name="auto_navi" pkg="rotors_gazebo" type="auto_navi" output="screen"/> 
    <node pkg="rostopic" type="rostopic" name="rostopic_dest" 
          args="pub  -1 /navigation_destination geometry_msgs/Point  -- 30 15 2"/>
    <node pkg="rostopic" type="rostopic" name="rostopic_restricted_area" 
          args="pub  -1 /Bounding_Area rotors_gazebo/restrictedSquare  --  11200 -140 -300  -200"/>
    
    <node name="hovering_example" pkg="rotors_gazebo" type="hovering_example" output="screen"/>
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
    <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher" />
  </group>

  
  <node pkg="octomap_server" type="octomap_tracking_server_node" name="octomap_talker" output="screen" args="$(arg path)">
    <param name="resolution" value="0.2" />
    <param name="frame_id" type="string" value="world" />
    <param name="sensor_model/max_range" value="10.0" />
    <param name="save_directory" value="$(optenv OCTOMAP_SAVE_DIR ./)" />
    <param name="track_changes" value="true"/>
    <param name="listen_changes" value="false"/>
    <param name="topic_changes" value="/octomap_tracking_server/changeset" />
    <param name="change_id_frame" value="$(arg changeIdFrame)" />
    <param name="min_change_pub" value="0" />
    <remap from="cloud_in" to="/firefly/vi_sensor/camera_depth/depth/points" />
  </node>

</launch>

`

gsilano, 2020-08-15:
Hmm. It is strange to me. Let me understand, are you able to run a simple simulation scenario with Crazyflie? I mean, one of those listed in the README file?

aka-LJ, 2020-08-20:
> Hmm. It is strange to me. Let me understand, are you able to run a simple simulation scenario with Crazyflie? I mean, one of those listed in the README file?

NO, I tried the mav_hovering_example.launch from rotors package, and the gazebo is not launching.

then I change the world to outdoor in crazyflie2_hovering_example.launch. gazebo is launching, but the world is not loading properly, and the drone is not hovering. stuck in the initial stage

closed 2020-09-17 by aka-LJ