gsilano/CrazyS #58: problem installing CrazyS in Ubuntu 18.04 using catkin_make

amijeet, 2020-09-15:
Hi, i am getting an error in hovering_example_spline_swarm.cpp file while using catkin_make after cloning.

[ 70%] Building CXX object CrazyS/rotors_control/CMakeFiles/crazyflie_complementary_filter.dir/src/library/crazyflie_complementary_filter.cpp.o
[ 70%] Linking CXX shared library /home/lethalwick/crazys_ws/devel/lib/libcrazyflie_complementary_filter.so
[ 70%] Built target crazyflie_complementary_filter
Scanning dependencies of target lee_position_controller
[ 70%] Building CXX object CrazyS/rotors_control/CMakeFiles/lee_position_controller.dir/src/library/lee_position_controller.cpp.o
[ 70%] Linking CXX shared library /home/lethalwick/crazys_ws/devel/lib/liblee_position_controller.so
[ 70%] Built target lee_position_controller
Scanning dependencies of target position_controller
[ 71%] Building CXX object CrazyS/rotors_control/CMakeFiles/position_controller.dir/src/library/position_controller.cpp.o
[ 71%] Linking CXX shared library /home/lethalwick/crazys_ws/devel/lib/libposition_controller.so
[ 71%] Built target position_controller
Scanning dependencies of target crazyflie_onboard_controller
[ 71%] Building CXX object CrazyS/rotors_control/CMakeFiles/crazyflie_onboard_controller.dir/src/library/crazyflie_onboard_controller.cpp.o
[ 72%] Linking CXX shared library /home/lethalwick/crazys_ws/devel/lib/libcrazyflie_onboard_controller.so
[ 72%] Built target crazyflie_onboard_controller
Scanning dependencies of target lee_position_controller_node
[ 73%] Building CXX object CrazyS/rotors_control/CMakeFiles/lee_position_controller_node.dir/src/nodes/lee_position_controller_node.cpp.o
[ 73%] Linking CXX executable /home/lethalwick/crazys_ws/devel/lib/rotors_control/lee_position_controller_node
[ 73%] Built target lee_position_controller_node
Scanning dependencies of target waypoint_publisher
[ 75%] Building CXX object CrazyS/rotors_gazebo/CMakeFiles/waypoint_publisher.dir/src/nodes/waypoint_publisher.cpp.o
[ 75%] Linking CXX executable /home/lethalwick/crazys_ws/devel/lib/rotors_gazebo/waypoint_publisher
[ 75%] Built target waypoint_publisher
Scanning dependencies of target hovering_example_spline_swarm
[ 76%] Building CXX object CrazyS/rotors_gazebo/CMakeFiles/hovering_example_spline_swarm.dir/src/nodes/hovering_example_spline_swarm.cpp.o
/home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp: In function ‘int main(int, char**)’:
/home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp:60:30: error: ‘DroneState’ is not a member of ‘mav_msgs’
       nh.advertise<mav_msgs::DroneState>(
                              ^~~~~~~~~~
/home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp:60:30: note: suggested alternative: ‘EigenDroneState’
       nh.advertise<mav_msgs::DroneState>(
                              ^~~~~~~~~~
                              EigenDroneState
/home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp:60:30: error: ‘DroneState’ is not a member of ‘mav_msgs’
/home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp:60:30: note: suggested alternative: ‘EigenDroneState’
       nh.advertise<mav_msgs::DroneState>(
                              ^~~~~~~~~~
                              EigenDroneState
/home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp:61:52: error: no matching function for call to ‘ros::NodeHandle::advertise<<expression error> >(const char [12], int)’
           mav_msgs::default_topics::DRONE_STATE, 10);
                                                    ^
In file included from /opt/ros/melodic/include/ros/ros.h:45:0,
                 from /home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/include/rotors_gazebo/parameters_ros.h:21,
                 from /home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp:23:
/opt/ros/melodic/include/ros/node_handle.h:249:15: note: candidate: template<class M> ros::Publisher ros::NodeHandle::advertise(const string&, uint32_t, bool)
     Publisher advertise(const std::string& topic, uint32_t queue_size, bool latch = false)
               ^~~~~~~~~
/opt/ros/melodic/include/ros/node_handle.h:249:15: note:   template argument deduction/substitution failed:
/home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp:61:52: error: template argument 1 is invalid
           mav_msgs::default_topics::DRONE_STATE, 10);
                                                    ^
In file included from /opt/ros/melodic/include/ros/ros.h:45:0,
                 from /home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/include/rotors_gazebo/parameters_ros.h:21,
                 from /home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp:23:
/opt/ros/melodic/include/ros/node_handle.h:315:13: note: candidate: template<class M> ros::Publisher ros::NodeHandle::advertise(const string&, uint32_t, const SubscriberStatusCallback&, const SubscriberStatusCallback&, const VoidConstPtr&, bool)
   Publisher advertise(const std::string& topic, uint32_t queue_size,
             ^~~~~~~~~
/opt/ros/melodic/include/ros/node_handle.h:315:13: note:   template argument deduction/substitution failed:
/home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp:61:52: error: template argument 1 is invalid
           mav_msgs::default_topics::DRONE_STATE, 10);
                                                    ^
/home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp:70:13: error: ‘DroneState’ is not a member of ‘mav_msgs’
   mav_msgs::DroneState trajectory_msg, trajectory_msg_pre;
             ^~~~~~~~~~
/home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp:70:13: note: suggested alternative: ‘EigenDroneState’
   mav_msgs::DroneState trajectory_msg, trajectory_msg_pre;
             ^~~~~~~~~~
             EigenDroneState
/home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp:71:3: error: ‘trajectory_msg’ was not declared in this scope
   trajectory_msg.header.stamp = ros::Time::now();
   ^~~~~~~~~~~~~~
/home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp:71:3: note: suggested alternative: ‘trajectory_pub’
   trajectory_msg.header.stamp = ros::Time::now();
   ^~~~~~~~~~~~~~
   trajectory_pub
/home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp:99:15: error: ‘eigenDroneFromStateToMsg’ is not a member of ‘mav_msgs’
     mav_msgs::eigenDroneFromStateToMsg(&eigen_reference, trajectory_msg);
               ^~~~~~~~~~~~~~~~~~~~~~~~
/home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp:99:15: note: suggested alternative: ‘EigenDroneStateDeque’
     mav_msgs::eigenDroneFromStateToMsg(&eigen_reference, trajectory_msg);
               ^~~~~~~~~~~~~~~~~~~~~~~~
               EigenDroneStateDeque
/home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp:106:7: error: ‘trajectory_msg_pre’ was not declared in this scope
       trajectory_msg_pre = trajectory_msg;
       ^~~~~~~~~~~~~~~~~~
/home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp:106:7: note: suggested alternative: ‘trajectory_pub’
       trajectory_msg_pre = trajectory_msg;
       ^~~~~~~~~~~~~~~~~~
       trajectory_pub
/home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp:120:30: error: ‘trajectory_msg_pre’ was not declared in this scope
       trajectory_pub.publish(trajectory_msg_pre);
                              ^~~~~~~~~~~~~~~~~~
/home/lethalwick/crazys_ws/src/CrazyS/rotors_gazebo/src/nodes/hovering_example_spline_swarm.cpp:120:30: note: suggested alternative: ‘trajectory_pub’
       trajectory_pub.publish(trajectory_msg_pre);
                              ^~~~~~~~~~~~~~~~~~
                              trajectory_pub
CrazyS/rotors_gazebo/CMakeFiles/hovering_example_spline_swarm.dir/build.make:62: recipe for target 'CrazyS/rotors_gazebo/CMakeFiles/hovering_example_spline_swarm.dir/src/nodes/hovering_example_spline_swarm.cpp.o' failed
make[2]: *** [CrazyS/rotors_gazebo/CMakeFiles/hovering_example_spline_swarm.dir/src/nodes/hovering_example_spline_swarm.cpp.o] Error 1
CMakeFiles/Makefile2:10051: recipe for target 'CrazyS/rotors_gazebo/CMakeFiles/hovering_example_spline_swarm.dir/all' failed
make[1]: *** [CrazyS/rotors_gazebo/CMakeFiles/hovering_example_spline_swarm.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
lethalwick@the:~/crazys_ws$ 

Could you please help me with this. thanks a lot.

gsilano, 2020-09-15:
Hi! Try to use `catkin build`, and let me know.

amijeet, 2020-09-15:
> Hi! Try to use `catkin build`, and then let me know.

I have used catkin_build several times, which throws a warning everytime and hangs.

gsilano, 2020-09-15:
Could you copy and paste the instructions you run?

amijeet, 2020-09-15:
i thoroughly followed your installation instructions on Ubuntu 18.04 with gazebo 9.
these were the instructions i have run several times. It hangs on the catkin build stage.
Thanks.

$ mkdir -p ~/catkin_ws/src
$ cd ~/catkin_ws/src
$ catkin_init_workspace  # initialize your catkin workspace
$ cd ~/catkin_ws/
$ catkin init
$ cd ~/catkin_ws/src
$ git clone -b dev/ros-melodic https://github.com/gsilano/CrazyS.git
$ git clone -b med18_gazebo9 https://github.com/gsilano/mav_comm.git
$ cd ~/catkin_ws

$ rosdep install --from-paths src -i
$ sudo apt install ros-melodic-rqt-rotors ros-melodic-rotors-comm ros-melodic-mav-msgs ros-melodic-rotors-control
$ sudo apt install ros-melodic-rotors-gazebo ros-melodic-rotors-evaluation ros-melodic-rotors-joy-interface
$ sudo apt install ros-melodic-rotors-gazebo-plugins ros-melodic-mav-planning-msgs ros-melodic-rotors-description ros-melodic-rotors-hil-interface
$ rosdep update
$ catkin build

gsilano, 2020-09-16:
Did you try to `catkin clean` before running the code? It seems strange because it works for me.

amijeet, 2020-09-16:
> Did you try to `catkin clean` before running the code? It seems strange because it works for me. Have you installed all the necessary packages?

I removed ros from my machine as well as reinstalled Ubuntu 18.04 and tried building crazyS from scratch. Exactly following your installation instructions. 
With catkin_make I get errors and the build fails as I mentioned previously. 
With catkin_build it gives me a bunch of cmake warnings. 
1st warning =>
Warnings << rotors_control:cmake /home/amijeet/catkin_ws/logs/rotors_control/butld.cnake.000.log CMake Warning at /opt/ros/melodic/share/cmake_modules/cmake/Modules/FindEigen.cmake:62 (message): The FindEigen.cmake Module in the cmake_modules package ts deprecated.

Please use the FindEigen3.cmake Module provided with Eigen. Change instances of find package(Eigen) to find_package(Eigen3). Check the FindEigen3.Cmake Module for the resulting CMake variable names.

Call Stack (most recent call first): CMakelists.txt:15 (find_package)

2nd warning => 
Warnings << rotors gazebo_plugins: Cmake /home/amijeet/catkin_ws/Logs/rotors gazebo_plugins/build.cmake.000. CMake Warning (dev) at Usr/share/cmake-3.10/Modules/FindBoost.cmake:911 (if): Poltcy CMPO054 is not set: Only interpret if() arguments as variables or keywords when unquoted. Run "Cmake - -help-policy CMPO054" for policy details. Use the cmake_policy command to set the policy and suppress this warning

Quoted variables like "chrono" will no longer be dereferenced when the policy is set to NEW. Since the policy is not set the OLD behavior will be used.

Call Stack (most recent call first): /usr/share/cmake-3.10/Modules/FindBoost.cmake:1558 (_Boost_MISSING_DEPENDENCIES)

Just/share/0GRE/cmake/modules/FindOGRE.cmake:318 (find_package)
 /usr/Lib/x86_64-linux-gnu/Cmake/gazebo/gazebo-config.cnake:175 (find_package) /opt/ros/melodic/share/gazebo_dev/cmake/gazebo_dev-extras.cmake:2 (find package)
 /opt/ros/melodic/share/gazebo dev/cmake/gazebo_devConftg.cmake:222 (include) CMakelists.txt:111 (find package) This warning is for project developers. Use -Wno-dev to suppress it.

3rd warning=>

CMake Warning at /opt/ros/melodic/share/catkin/cnake/catkin_package.cmake:166 (message):
catkin_package() DEPENDS on 'octomap' but neither 'octomap INCLUDE_DIRS' nor 'octomap_LIBRARIES' is defined.
call stack (most recent call first): /opt/ros/melodic/share/catkin/cmake/catkin_package.cnake:102 (_catktn_package) CMakelists.txt:223 (catkin_package)

4th warning=> 

CMake Warning at /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:166 (message): catkin_package() DEPENDS on 'opencv' but neither 'opencv_INCLUDE DIRS' nor opencv LIBRARIES' is defined. Call Stack (most recent call first):

/opt/ros/melodic/share/catkin/cmake/catktn_package.cnake:102 (catkin_package) CMakelists.txt:223 (catkin_package)

Build says 11/13 complete and my machine freezes here, mouse doesn't move, takes no keyboard input, I have to forcefully reboot my machine. 
Also I have tried doing this multiple times. But this is what happens everytime. I am very thankful for the time you have been giving to my problem, just hope I can install and build CrazyS once and continue with my research. 
Thanks a lot

gsilano, 2020-09-16:
Warnings are fine, less so if the machine freezes. How many gigabytes of RAM do you have?

amijeet, 2020-09-16:
> Warnings are fine, less so if the machine freezes. How many gigabytes of RAM do you have?

I got 4 gigabytes of RAM

gsilano, 2020-09-16:
Are not enough to compile the code. You need at least 12GB.

amijeet, 2020-09-16:
> Are not enough to compile the code. You need at least 12GB.

But it worked for a friend of mine with the same specifications as mine. Even he's got 4 gigabytes of RAM. It ran once and he even did the hover simulation, after that, he gets the same warnings and his laptop freezes. 
Sometimes it says build error.

gsilano, 2020-09-16:
This is my workaround. In the following you find the links to download the virtual machines I made for this type of errors. 

[Ubuntu 16.04 with ROS Kinetic and Gazebo 7](https://mega.nz/#!oxdwzaIY!C27ekXwbObbiU37BrVQSjeRMP2HCACVgQj1WbvVmXlA)

[Ubuntu 16.04 with ROS Kinetic and Gazebo 9](https://mega.nz/#!550ASCrZ!JYm1mFFlYa9fNh_Cb4Z6p7l9k6Vu5tpRJB3NjMoSF1A)

amijeet, 2020-09-17:
> This is my workaround. In the following you find the links to download the virtual machines I made for this type of errors.
>
> [Ubuntu 16.04 with ROS Kinetic and Gazebo [...]

Okay, I shall try those. Thanks a lot again

closed 2020-11-21 by gsilano