gsilano/CrazyS #59: changing roll pitch and yaw of the drone

amijeet, 2020-10-19:
Hi Giuseppe
Sorry to trouble you again, but is there a way to change the roll, pitch and yaw of the drone in the **hovering example** without making making changes in the final point where the crazyflie tries to reach ?
I want it to reach a point, but i want to send random roll, pitch and yaw values while it is trying to reach its final destination and test my reinforcement learning algorithm there. I have tried my algorithm on various drones, but testing it here is a little difficult for me. I want some degree of randomness on the flight path (if not randomness, some deviation while it tries to reach a point in space) in the hovering example.

gsilano, 2020-10-20:
Hi @amijeet! The current implementation of the control algorithm, even considering the latest additions (see CHANGELOG file), does not allow you to specify the roll, pitch, and yaw angles. You should design your own algorithm and then implement it in ROS. You could start with those available on [RotorS](https://github.com/ethz-asl/rotors_simulator), on [Hector Quadrotor](https://github.com/tu-darmstadt-ros-pkg/hector_quadrotor), and on [MRS Platform](https://github.com/ctu-mrs/uav_core).
I don't currently plan to work on this project. In the future, I hope near, I could resume work and expand it.

gsilano, 2020-12-24:
Hi @amijeet ! Can I close this issue?

amijeet, 2020-12-24:
Yeah sure. 
Thank you so much for being so helpful ❤️

closed 2020-12-24 by gsilano