gsilano/CrazyS #60: ratio between motor speed and thrust

XuboLyu, 2020-12-02:
Hi @gsilano @lianne @growupboron @EmanueleAucone 

Thank you so much for your work. It helps a lot for crazyflie simulation!

Recently I was working on transferring the trained model from other simulation to the crazyS. However, I was wondering where I can find the ratio or maybe a  formula F such that I can use it to do the transformation "motor_speed = F(thrust)". Which conversion formula are you using in CrazyS? 

Thanks!

gsilano, 2020-12-03:
Hi @XuboLyu! Have you seen the paper reported on [this page](https://github.com/gsilano/CrazyS/wiki/Publications)?

XuboLyu, 2020-12-09:
@gsilano 

Thanks for pointing the papers. I'll take a look at them.

closed 2020-12-09 by XuboLyu