gsilano/CrazyS #61: ANGULAR_MOTOR_COEFFICIENT inconsistency with the publication

OAHC2022, 2021-01-24:
Hello,

When I was reading the [paper](https://giuseppesilano.net/publications/rosChapter19.pdf). I came across the equation:

<p align="center">
<a href="https://www.codecogs.com/eqnedit.php?latex=\omega_i=\frac{\pi}{30}(\alpha&space;*&space;\text{PWM}_i&space;&plus;&space;q)" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\omega_i=\frac{\pi}{30}(\alpha&space;*&space;\text{PWM}_i&space;&plus;&space;q)" title="\omega_i=\frac{\pi}{30}(\alpha * \text{PWM}_i + q)" /></a>
</p>

However, when I checked the code in [here](https://github.com/gsilano/CrazyS/blob/master/rotors_control/src/library/position_controller.cpp#L268). I realized that there was inconsistency with the equations. Given the values defined in the macros, the equation in the code was

<p align="center">
<a href="https://www.codecogs.com/eqnedit.php?latex=\omega_i=\alpha&space;*&space;\text{PWM}_i&space;&plus;&space;\frac{\pi}{30}&space;*&space;q" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\omega_i=\alpha&space;*&space;\text{PWM}_i&space;&plus;&space;\frac{\pi}{30}&space;*&space;q" title="\omega_i=\alpha * \text{PWM}_i + \frac{\pi}{30} * q" /></a>
</p>

I wonder which one should I use. 

Thanks

gsilano, 2021-01-24:
Hi @OAHC2022! I checked out the code. The formula at line 268 of the `position_controller.cpp` file is the following

```
\omega_i = \alpha PWM_i + q
```
where `q` is expressed in rad/s as `PWM_i` and `\omega_i`. I can understand this could be misleading as PWMs are usually measured in rounds per second and not radians per second. However, Gazebo uses the SI unit measure to express all the data. Therefore, position, linear and angular velocities are expressed in `m`, `m/s` and `rad/s`, respectively. For this reason, the term `\pi/30` need not be taken into account.
In the paper, we preferred to refer to the same nomenclature of [43](https://arxiv.org/pdf/1608.05786)

If you take a look at the whole chain (i.e., RateController, ControlMixer, etc.), you will simply understand what I mean. If you have any other questions, please do not hesitate.

OAHC2022, 2021-01-24:
Oh I see the change in measurement indeed is tricky. 

Thank you.

closed 2021-01-24 by OAHC2022