gsilano/CrazyS #64: My Simple Questions

Tevhit, 2021-03-23:
Hello Giuseppe, I have two question. 

When I launch the "crazyflie2_swarm_hovering_example.launch" file, the UAV's take altitude then they wait. Although I published right-left and up-down movement topics, they don't move any. I only want that I will publish to take altitude topic, then the UAV will take altitude. How can I do that? Are there any documentation for beginners? Are there any documentation that explaining all topics and all nodes?

Also, I want to use python while controlling UAV's. Is it possible?

I hope you help, best, 
Tevhit.

gsilano, 2021-03-24:
Hi! As you can see from #44, the swarm example was built to work with multiple quad-rotors. The supplied controller does not perform any trajectory tracking. This is why you are unable to move quad-rotors in `x`, `y`, and `z` space whatever the inputs are.

Yes you can write your own ROS node in Python. Please, refer to [this guide](http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28python%29)

I suggest you take a look at the related [Publications](https://github.com/gsilano/CrazyS/wiki/Publications) page.

Tevhit, 2021-03-28:
Thansk

closed 2021-03-28 by Tevhit