gsilano/CrazyS #68: Need help for identifying files for spawning drone at an elevated platform and flight pathing

Wizzya, 2021-04-09:
Hello, i am quite new to drone programming and would like some assistance requiring the editing and spawning of drones. I have a school project that requires me to spawn the drones and make it fly in formation within the simulated environment. I am currently at the part where i managed to spawn the drone in the yosemite environment. I am using the crazyfile2_swarm hovering_example launch However i am having difficulties in 3 areas:
1. Spawning the drone at an elevated height, probably as i could not find the file that controls the spawning of the drones.
2. Identifying the files that helps the drones to fly forward, backward and to disable the way point
3. Environment spawn for the drone is always on the flat ground. I have seen some tutorials that allow drones to spawn on hills, uneven surfaces with coordinates. 
1a. I have tried to mess with the drone spline trajectory and at set initial position in the swarm file, however the drone does not react to the changes in value of z and only affects the spawn position along the x and y axis. From my understanding spline affects the final position and velocity, acceleration of the drone.
2a. Was assuming that drone has an integrated flight path and tried to disable the internal model controller but ended up messing up the program completely, or the drone file just launches as default in swarm hovering example
3a. Tried to find coordinates to work around but have no idea how to do it or start with. :(

Thanks for the help in adavnce.

gsilano, 2021-04-10:
Hello! As discussed in the latest issues (#64, #67, etc.), the available controller does not allow for any trajectory tracking and formation flights. So, if you want to use the current configuration, you need to design your own control algorithm. This answers the first two points of your question.

If you want to fly in formation and are not interested in Crazyflie (your goal is to fly with a quad-rotor), I can recommend you use [this repository](https://github.com/ctu-mrs/mrs_uav_system). There you will find all the information you need. Also, you can use the issue tracker for direct questions.

I hope this helps you. If you have any further questions, please do not hesitate.

Wizzya, 2021-04-12:
My project requires me to use crazyfile 2.0 nanodrones for hardware integration, so in order to do control algorithm at at collage level, where would you recommend i start with? Also, do i have to replace the written code for the file  "internal model control"  so that i am able to spawn the drone at varying heights (eg. on the hills) in the created gazebo worlds.

gsilano, 2021-04-13:
The controller available in the repository does not allow any trajectory tracking. This means that if you would like to track a given trajectory, you have to design your own controller. The controller works only in the scenario in which the drone starts from the ground.
As for your second question, you can spawn the Crazyflie at a given altitude. To do that, simply modify the `x`, `y`, and `z` variables in the `spawn_mav_crazyflie.launch' file. For simplicity, I suggest you follow what I did from line 22 to integrate these variables in the `crazyflie2_hovering_example.launch` or any other launch file you wish to use.

I hope this helps.

closed 2021-04-15 by Wizzya