gsilano/CrazyS #69: crazyflie2_swarm_hovering_example.launch runs but does not hover and can not be controlled.

terilgen, 2021-04-23:
We use Ubuntu 18.04 and ROS Melodic. When we try launch the simulation all the quadcopters are visible but the topics to control them do not start. We also tried to control them with xbox360 controller but it did not work as well.

gsilano, 2021-04-24:
Hi @terilgen! This is a common problem when using the package with Ubuntu 18.04 and Gazebo 9. I highly recommend you follow what has been reported in other issues of this repository. The problem has been addressed many times, and quite sure that you will find the solution. If not, we can see together what is wrong with it.

terilgen, 2021-04-30:
Hello again sir @gsilano , sorry for the late reply. I was dealing with my midterms. I have looked those issues and tried what they did but it did not work. When i use your VM file it works very well all topics are can be seen but when i try it on my computer with (Ubuntu 18.04 + ROS Melodic + Gazebo9) most of the important topics can not be seen. I added my errors and topic pictures on the attachment. 

This is from Ubuntu 18.04
[image]

[image]

[image]

[image]

And this one is from Virtual Machine (Ubuntu 16.04)
[image]

gsilano, 2021-05-02:
I see your problem. Again, I think this is related to a wrong symbolic link. Have you checked the content of the `bashrc` file? As a quick test, you could compare the content of the file with that in the VM. That way, you can understand if something is missing there. Also, do not forget to source the file to the terminal line after any modifications.

Next time, please, do not post photos but try to attach to the issue the log file. Thanks!

closed 2021-05-02 by terilgen