gsilano/CrazyS #70: the problem of crazyflie2 's config parmas

Suvian-wy, 2021-05-10:
Firstly, thanks for ding so many wroks! I have a simple quenstion.
I don't know what't the mean of the param "angle" in files crazyflie2.yaml.  Then if the -3.14 and 3.14 is a same angle or not?

[image]

gsilano, 2021-05-10:
Hi! As described in [RotorS – A Modular Gazebo MAV Simulator Framework](https://www.researchgate.net/publication/309291237_RotorS_-_A_Modular_Gazebo_MAV_Simulator_Framework) and in [CrazyS: a software-in-the-loop simulation platform for the Crazyflie 2.0 nano-quadcopter](https://giuseppesilano.net/publications/rosChapter19.pdf), the "angle" variable describes the angles spanned by each rotor starting from the origin of the body frame (it is located in the middle of the drone). Yes, you are right `-3.14` and `+3.14` are the same angle, but the sign identifies the direction. So whether they are positioned counter-clockwise or clockwise.

Suvian-wy, 2021-05-14:
yeah, I know they have differet directio, but  if the two rotors will in a same position?

gsilano, 2021-05-14:
What does "if the rotors have the same position" mean? I think you are confusing the spinning direction of the rotors (clockwise or counter-clockwise) with their position in the XY-plane. Let's think about the rotors as a point moving on a circle. So, how we can describe its position? We can provide its x and y projects on the XY reference frame (Cartesian frame) or we can use polar coordinates, i.e., the arm length and the angle spanned by the rotor from the origin of the reference frame to its position in the XY plane. 

I hope it is clearer now than before.

Suvian-wy, 2021-05-14:
ok,thanks, maybe I have no clear idea in the spinning rotors. But this confusing have no  influence in using the project. So I think I can close this issue.

Thank you!

closed 2021-05-14 by Suvian-wy