gsilano/CrazyS #71: Writing low-level Controller

sunnyshi0310, 2021-05-14:
Hi, 
I am working on writing a low-level controller (like the position controller or the IMC) for the robot. I try to publish the message of type "Actuators" to the topic '/crazyflie2/command/motor_speed' while I am not sure how to set the "angles" and 'angular_velocities' in that message. Could you please introduce more on how to drive the motor of the robot? Thanks in advance.

Suvian-wy, 2021-05-14:
only need to set the `angluar_velocities` with the motor reference velocities  as a (1x4) array, the rotor will work!

sunnyshi0310, 2021-05-14:
Hi, @Suvian-wy Thanks for your reply. I can successfully drive the motor while do you know the physical meaning of the angular_velocities, in other words, do you have any idea on how to calculate the angular velocities from the four force F applied?

Pooyanrmz, 2021-05-16:
Hi @sunnyshi0310, angular velocities depend on the dynamics of your model and the power of its brushless dc motors. However, in the case of most quadrotors, the thrust force of a single rotor is defined by multiplying a thrust constant by the square of rotors' angular velocity.

closed 2021-09-03 by stale[bot]