gsilano/CrazyS #73: Question: Is it possible to link this project with UCT-ASTlab Crazyswarm?

santiagorg2401, 2021-05-18:
Hello, I recently installed your project, I've been working with Postdoc Hönig's Crazyswarm project for several months but I'd like to simulate on a Gazebo world, my professor recommended me your work, so, is there a way to link it with Hönig's (the Crazyflie python class) or do I have to write a ROS Node in order to send the desired coordinates to the virtual Crazyswarm on Gazebo?

Thank you in advance.

gsilano, 2021-05-18:
Hi! Nice to hear from you that you have found my work of interest for your research. As for your question, honestly, I don't know how `Crazyswarm` works. I mean, what is its I/O interface, and how the package handles the communication with the nanoquadrotor. I'm aware that others have already tried to make it works (e.g, [crazyflie_nmpc](https://github.com/bcbarbara/crazyflie_nmpc)). So, I would recommend you to start from the [references papers in the Wiki](https://github.com/gsilano/CrazyS/wiki/Publications), and then give me back with your questions.

santiagorg2401, 2021-05-19:
Understood, thank you.

closed 2021-05-23 by gsilano