gsilano/CrazyS #76: Virtual Reality Integration :help wanted:

muhammete1, 2021-07-14:
Hi There,

First of all, thanks for sharing this project. We have a research project about trying to control drones with swarm behavior. We don't want to use Crazyswarm while doing this. Currently, We are doing a test with Crazyflie 2.1. How can we apply virtual reality integration with CrazyS or is there another easier way to do it ?

gsilano, 2021-07-15:
Hi @muhammete1! What do you mean by "swarm behavior"?
Well, I would say, if you have your swarm algorithm in MATLAB or any other place (e.g., Python or C++ code), you can translate it into ROS or you can use the Robotic System Toolbox to connect MATLAB in the loop with Gazebo (see [this guide](https://github.com/gsilano/CrazyS/wiki/Interfacing-CrazyS-through-MATLAB)). Then, you can test your algorithm by taking advantage of Software-in-the-loop simulations. [This paper](https://ieeexplore.ieee.org/document/8914154) may help you understand what I mean.

I hope this helps.

gsilano, 2022-04-16:
What's reported in this issue could be of help https://github.com/ethz-asl/rotors_simulator/issues/487#issuecomment-1100565187. Now, this is part of the Wiki. This has been not tested yet.

closed 2022-04-16 by gsilano