gsilano/CrazyS #77: ROS GAZEBO

hazarD77, 2021-07-15:
Hi, i work on Gazebo with crazyflie and i succeed to open crazyflie files on ROS, the crazyfile is flying but
I couldn't find the file where I can give the target location while the drones are in the air. I want to show target point to drones. How i succeed that?
Thanks...
Best Regards.

gsilano, 2021-07-15:
Please, read #67, #65, #44, #55. If you have other questions, just drop me a message.

hazarD77, 2021-07-21:
Hi, sir
I read and research the subject headings in the #67, #65, #44, #55. Therefore, i want to ask some question. If you answer me, i am glad.
1) How to transfer our own algorithms to CrazyS files?
2) How can I achieve this when I want to integrate virtual and real?
3) Do I need to use a Gazebo environment for Crazyflie Virtual Reality Flight, can I use another simulation program?
Thanks.
Best Regards...

gsilano, 2021-07-21:
My answers are inline:

1. What do you mean by your own algorithm? Is it a controller? If so, you need to write your `C++` file and include it in the `rotors_control/src/library` folder.  Next, you need to edit the `position_controller_node.cpp` file to call your controller, and not the standard one in CrazyS. Of course, you have to follow the flow understanding if other changes are required. The line is reported below for your convenience. 
 https://github.com/gsilano/CrazyS/blob/fa363088f3eb96d77b961f5c6cd6e5cc924400bb/rotors_control/src/nodes/position_controller_node.cpp#L502

Finally, you need to include the code in the `CMakeList.txt` file. Just follow the file for the other libraries, i.e., `crazyflie_onboard_controller.cpp`, `crazyflie_complementary_filter.cpp`, and so on.

2. If you want to fly with the Crazyflie, you can take a look at these two repositories: [crazyflie_nmpc](https://github.com/bcbarbara/crazyflie_nmpc) and [crazyflie_cpp](https://github.com/whoenig/crazyflie_cpp). As far as I know, they are the best to cope with this task.

3. CrazyS was designed to work with Gazebo. I don't know which robotic simulator you have in mind.  However, whatever the simulator, the code requires some changes.

gsilano, 2022-04-16:
The controller #86 is not ready to be integrated yet. Issue closed for inactivity.

closed 2022-04-16 by gsilano