gsilano/CrazyS #78: Virtual Reality Integration

muhammete1, 2021-07-24:
Can we provide virtual-reality integration with using PWM datas which coming from Crazyflies? If it is not possible, is there any way to do this?

gsilano, 2021-07-25:
Hi! If I get it correctly, you would like to use ROS, and therefore CrazyS, in the loop with the real device. Isn't it? 

Well, firstly, why do you want to use CrazyS to control the Crazyflie? [Crazyflie ROS](https://github.com/whoenig/crazyflie_ros) is the most suitable package in this case. I would use MATLAB first, or any other numerical environment to test the path planning/control algorithm, then CrazyS (if you want to run a more advanced simulation that takes into account second-order effects as well as the onboard firmware), and then the real device. 

Please note that CrazyS does not integrate the latest version of the firmware (it is still at the `2018.01` release). Therefore, you should check if the main functions, i.e., controller and state estimator, have been modified in the meantime.

I hope this helps.

muhammete1, 2021-07-26:
Hi! Yes, we are using Crazyswarm and we using drones for making variable formations. We want to give PWM variables and use them on CrazyS for virtual-reality environment. In other words, We want to simulate simultaneous on Crazys the commands that on impements real Crazyflies.

We already have tested algorithms on numerical and real environments. But we don't able to control our drones on CrazyS. We just want to operate virtual(CrazyS) and real devices simultaneously.

Additionally, we want to remind you that we are using Crazyswarm these all our stuides.

Thanks in advance.

gsilano, 2021-07-26:
I see! Please, note that Gazebo only accepts SI measurements (i.e., meters per second, radians per second, etc.). Therefore, you need to convert PWM signals from propeller angular speeds expressed in radians per second. If you see the reference publications (see the book chapter in the Wiki), you will easily understand how to convert these values.

So, you need to write some glue code to interface Gazebo with your code. [Here](https://github.com/gsilano/CrazyS/blob/master/rotors_gazebo/launch/crazyflie2_without_controller.launch), there is a simple launch file that runs the Crazyflie without any controller.

muhammete1, 2021-07-27:
Thanks for your quick respond!

How can we apply this converted data to Crazyflie with Gazebo? How can we create our own controller? Is there any source that you can advice about this controller situation?

Thanks in advance.

gsilano, 2021-07-27:
Well, as I said, it simply implements the equation referenced in the book chapter. As for the controller, you can use the book chapter as a guideline to implement the position controller (see the `position_controller_node.cpp` and `position_controller.cpp` as examples)

muhammete1, 2021-07-30:
Another question is how can we update version 2018.01 to version 2020.02? Is there any reference about it? Thanks in advance.

gsilano, 2021-07-31:
You need to explore the firmware of the Crazyflie [available in this repository](https://github.com/bitcraze/crazyflie-firmware). Next, compare the onboard controllers in CrazyS (i.e., `crazyflie_complementary_filter.cpp` and `sensfusion6.cpp` files) looking for differences/updates.

semihsahhinx, 2021-08-05:
position_controller_node.cpp ==>   512 ==>    actuator_msg->header.stamp = odometry_msg->header.stamp; 
Here, we changed values that sent to rotor. Below this line (512) we tried these values which taken by real drone but drone is not flying with these values only the propellers are running.
actuator_msg->angular_velocities[0] = 2000;
actuator_msg->angular_velocities[1] = 2000;
actuator_msg->angular_velocities[2] = 2000;
actuator_msg->angular_velocities[3] = 2000;
How can I solve this problem?

gsilano, 2021-08-06:
You need to change the values in [this file](https://github.com/gsilano/CrazyS/blob/master/rotors_description/urdf/crazyflie2.xacro): `crazyflie.xacro`. Please, before doing this, read #56

semihsahhinx, 2021-08-06:
> I see! Please, note that Gazebo only accepts SI measurements (i.e., meters per second, radians per second, etc.). Therefore, you need to convert PWM signals from propeller angular speeds expressed [...]
-------------------------
As you mentioned, we should convert these values. But I couldn't find in book chapter. Can you please help me to convert these values?

gsilano, 2021-08-06:
See page 9 in the [book chapter](https://giuseppesilano.net/publications/rosChapter19.pdf)

gsilano, 2022-04-16:
Closed for inactivity.

closed 2022-04-16 by gsilano