gsilano/CrazyS #79: Cannot make it work on Melodic

barbosafs, 2021-08-18:
Hello!
I've been trying to get CrazyS to work on my machine for a couple of days now, but with no success. I have an Ubuntu 18 with ROS Melodic and Gazebo 9. I followed all the corresponding installation steps listed [here](https://github.com/gsilano/CrazyS#installation-instructions---ubuntu-1804-with-ros-melodic-and-gazebo-9).

After running `rosdep check --from-paths src` I get the following:
```
All system dependencies have been satisfied
ERROR[mav_comm]: Cannot locate rosdep definition for [mav_system_msgs]
	rosdep key : mav_system_msgs
	OS name    : ubuntu
	OS version : bionic
	Data:
<no data>
```

The following is the verbose of `catkin build`:
```
--------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/melodic
Workspace:                   /home/fdsb/crazy_ws
--------------------------------------------------------
Build Space:        [exists] /home/fdsb/crazy_ws/build
Devel Space:        [exists] /home/fdsb/crazy_ws/devel
Install Space:      [unused] /home/fdsb/crazy_ws/install
Log Space:         [missing] /home/fdsb/crazy_ws/logs
Source Space:       [exists] /home/fdsb/crazy_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
--------------------------------------------------------
[build] Found '14' packages in 0.0 seconds.                                    
[build] Updating package table.                                                
Starting  >>> catkin_tools_prebuild                                            
Finished  <<< catkin_tools_prebuild                 [ 0.9 seconds ]            
Starting  >>> mav_msgs                                                         
Starting  >>> mav_state_machine_msgs                                           
Starting  >>> mav_system_msgs                                                  
Starting  >>> rotors_comm                                                      
Starting  >>> rotors_description                                               
Starting  >>> rotors_evaluation                                                
Starting  >>> rqt_rotors                                                       
Finished  <<< rotors_description                    [ 1.5 seconds ]            
Finished  <<< rqt_rotors                            [ 2.3 seconds ]            
Finished  <<< rotors_evaluation                     [ 2.3 seconds ]            
Finished  <<< mav_state_machine_msgs                [ 3.9 seconds ]            
Finished  <<< mav_system_msgs                       [ 3.9 seconds ]            
Finished  <<< rotors_comm                           [ 4.7 seconds ]            
Finished  <<< mav_msgs                              [ 5.8 seconds ]            
Starting  >>> mav_planning_msgs                                                
Starting  >>> rotors_control                                                   
Starting  >>> rotors_joy_interface                                             
_______________________________________________________________________________
Warnings   << rotors_control:cmake /home/fdsb/crazy_ws/logs/rotors_control/build.cmake.000.log
CMake Warning at /opt/ros/melodic/share/cmake_modules/cmake/Modules/FindEigen.cmake:62 (message):
  The FindEigen.cmake Module in the cmake_modules package is deprecated.

[... 234 lines cut ...]
[ INFO] [1629276401.828865126]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1629276401.897418615]: Physics dynamic reconfigure ready.
[INFO] [1629276402.013537, 0.000000]: Calling service /gazebo/spawn_urdf_model
[ INFO] [1629276402.364748662]: Unpaused the Gazebo simulation.
[INFO] [1629276402.364747, 0.000000]: Spawn status: SpawnModel: Successfully spawned entity
gzserver: symbol lookup error: /opt/ros/melodic/share/gazebo_plugins/../../lib/librotors_gazebo_multirotor_base_plugin.so: undefined symbol: _ZN14gz_sensor_msgs9ActuatorsC1Ev
[gazebo-2] process has died [pid 25673, exit code 127, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -u -e ode /home/fdsb/crazy_ws/src/CrazyS/rotors_gazebo/worlds/basic_crazyflie.world __name:=gazebo __log:=/home/fdsb/.ros/log/cda2315c-0000-11ec-a4e3-9cb6d01a82c1/gazebo-2.log].
log file: /home/fdsb/.ros/log/cda2315c-0000-11ec-a4e3-9cb6d01a82c1/gazebo-2*.log
[crazyflie2/spawn_crazyflie2-4] process has finished cleanly
log file: /home/fdsb/.ros/log/cda2315c-0000-11ec-a4e3-9cb6d01a82c1/crazyflie2-spawn_crazyflie2-4*.log
^C[crazyflie2/quaternion_to_rpy-9] killing on exit
[crazyflie2/joint_state_publisher-8] killing on exit
[crazyflie2/robot_state_publisher-7] killing on exit
[crazyflie2/hovering_example-6] killing on exit
[crazyflie2/position_controller_node-5] killing on exit
[gazebo_gui-3] killing on exit
[gazebo_gui-3] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Any suggestions on how to fix this?

gsilano, 2021-08-18:
Hello! Sorry, I didn't get it from the issue. Were you able to build the code?

The error appears in this line:

```
gzserver: symbol lookup error: /opt/ros/melodic/share/gazebo_plugins/../../lib/librotors_gazebo_multirotor_base_plugin.so: undefined symbol: _ZN14gz_sensor_msgs9ActuatorsC1Ev
```

I think you have not installed the package correctly. Please do the following lines again

```
## Environments instructions
sudo apt-get update -qq
sudo apt-get install dpkg
sudo rm /var/lib/dpkg/lock
sudo dpkg --configure -a
sudo apt install -y python-catkin-tools python-catkin-pkg python-rosdep python-wstool ros-melodic-desktop-full 
sudo apt-get install -y ros-melodic-joy  python-rosinstall python-rosinstall-generator python-wstool build-essential
sudo apt install -y ros-melodic-octomap-ros ros-melodic-mavlink python-rosinstall build-essential protobuf-compiler
sudo apt install -y libgoogle-glog-dev python-rosinstall-generator ros-melodic-pointcloud-to-laserscan
sudo apt install -y python-rosdep ros-melodic-octomap ros-melodic-octomap-mapping
sudo apt-get install -y ros-melodic-joy ros-melodic-control-toolbox
source /opt/ros/melodic/setup.bash
sudo rosdep init
rosdep update --include-eol-distros
rosdep install octomap_mapping
rosmake octomap_mapping

## CrazyS instructions
source /opt/ros/melodic/setup.bash
mkdir -p ~/catkin_ws/src
catkin_init_workspace
catkin init
cd ~/catkin_ws/src
git clone -b dev/ros-melodic https://github.com/gsilano/CrazyS.git
git clone -b med18_gazebo9 https://github.com/gsilano/mav_comm.git
rosdep update
cd ~/catkin_ws
rosdep install --from-paths src -i
rosdep update
catkin build
echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
source ~/.bashrc
```

I hope this helps.

barbosafs, 2021-08-18:
Worked like a charm! Thank you so much mate.
Something in these lines did the job:
```
sudo apt-get update -qq
sudo apt-get install dpkg
sudo rm /var/lib/dpkg/lock
sudo dpkg --configure -a
sudo apt install -y python-catkin-tools python-catkin-pkg python-rosdep python-wstool ros-melodic-desktop-full 
```

closed 2021-08-18 by barbosafs