gsilano/CrazyS #80: Some issues on installation in ubuntu 20.04

Yiqing-Gu, 2021-09-08:
Hi,

I am working on installation of CrazyS in Ubuntu 20.04. I met two problems.

1. When I ran "rosdep install --from-paths src -i", I got an error "ERROR: the following packages/stacks could not have their rosdep keys resolved to system dependencies: rotors_simulator: Cannot locate rosdep definition for [rotors_hil_interface]". It seems that I do not have such a folder called "rotors_hil_interface". 

2. In part 3 "Build your workspace with python_catkin_tools", I tried to install all packages like "ros-melodic-rqt-rotors". But I got "Unable to locate package ros-melodic-rqt-rotors". Even I used "ros-noetic-rqt-rotors" and such a package does exist. However, I still got same error. I could not install all packages in part 3.

About python_catkin_tools, it seems that "sudo apt install python-catkin-tools" does not work in Ubuntu 20.04. I used "sudo apt install python3-catkin-tools" instead.

Looking for any answers. Thank you very much!

gsilano, 2021-09-09:
I think there was an error in the installation procedure. Did you install CrazyS from the `dev / ros-melodic` branch? Even if the name says "ROS Melodic", it should also be compatible with Noetic. At least I hope so.

Obviously, if you are running Ubuntu 20.04 with ROS Noetic, you cannot install the `melodic` packages or simply change the name of the ROS distribution. Below are the instructions to follow:

```
$ mkdir -p ~/catkin_ws/src
$ cd ~/catkin_ws/src
$ catkin_init_workspace  # initialize your catkin workspace
$ cd ~/catkin_ws/
$ catkin init
$ cd ~/catkin_ws/src
$ git clone -b dev/ros-melodic https://github.com/gsilano/CrazyS.git
$ git clone -b med18_gazebo9 https://github.com/gsilano/mav_comm.git
$ cd ~/catkin_ws
$ rosdep update
$ catkin build
$ echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
$ source ~/.bashrc
```

Hopefully, using the command `sudo apt install python3-catkin-tools` instead of` sudo apt install python-catkin-tools` shouldn't cause any problems. However, I cannot guarantee anything as it has never been tested with this ROS distro.

I hope this helps.

fabrizzio-gz, 2021-11-15:
I'm having the same issue. After running `rosdep install --from-paths src -i`, I got the following error message.
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
rotors_simulator: Cannot locate rosdep definition for [rotors_hil_interface]
```
I tried installing all the packages for the Noetic version (Ubuntu 20.0.4). As far as I can tell, the problem is that the `rotors_simulator` package can run only on Melodic ([http://wiki.ros.org/rotors_simulator](http://wiki.ros.org/rotors_simulator))

gsilano, 2021-11-17:
As far as I know, RotorS, the drone simulator used as the basis for CrazyS, does not work with ROS Noetic. In this case, I would rather suggest opening an issue on the [RotorS issue tracker](https://github.com/ethz-asl/rotors_simulator/issues). So if successful, please let us know. I will try to create a version of CrazyS for this version of ROS.

fabrizzio-gz, 2021-11-17:
Apparently, RotorS has added support for Noetic ([merged pull](https://github.com/ethz-asl/rotors_simulator/pull/651)). However, I was unable to run it along CrazyS. I'll let you know if I find a workaround to this issue.

Yiqing-Gu, 2021-11-18:
You can try this CrazyS. It may work for you. https://github.com/caslabuiowa/CrazyS
You need to clone its dev/ros-melodic branch.

If not, try to install the following dependencies first.
`sudo apt install python3-rosdep python3-rosinstall python3-rosinstall-generator python3-wstool build-essential

sudo apt-get install ros-noetic-octomap ros-noetic-octomap-mapping ros-noetic-octomap-rviz-plugins ros-noetic-octomap-server ros-noetic-octomap-msgs ros-noetic-joy ros-noetic-geodesy ros-noetic-octomap-ros ros-noetic-mavlink ros-noetic-control-toolbox

sudo apt-get install libgflags-dev

sudo apt install libgoogle-glog-dev

sudo apt-get install protobuf-compiler libprotobuf-dev`

gsilano, 2021-11-27:
Hi there! This Christmas holidays, I will be working on CrazyS to make it ROS Neotic compatible. I will reale a new version as soon as it is ready.

fabrizzio-gz, 2021-11-29:
> Hi there! This Christmas holidays, I will be working on CrazyS to make it ROS Neotic compatible. I will reale a new version as soon as it is ready.

Great news! Thank you for your dedication and hard work.

closed 2021-12-09 by Yiqing-Gu

gsilano, 2021-12-28:
CrazyS is now available for ROS Noetic and Gazebo 11 on Ubuntu 20.04 (#89)