gsilano/CrazyS #82: Adding a custom MAV

dkapur17, 2021-10-01:
I have the `.xacro` file of a custom MAV, that I need to simulate in Gazebo. The README says that "The contribution can be considered as a reference guide for expanding the RotorS functionalities in the Unmanned Aerial Vehicles (UAVs) field, by facilitating the integration of new aircraft."

Unfortunately, I have not been able to find any guides on integrating a new aircraft into the package. Could you please point me in the right direction? Thanks a lot!

gsilano, 2021-10-02:
Hi there! Have you read the [reference publications](https://github.com/gsilano/CrazyS/wiki/Publications)? I highly recommend the book chapter. There you can find a detailed description of how the Crazyflie was integrated into the RotorS simulator.

I hope this helps. Please don't hesitate to give me back if this doesn't work for you.

michieltebraake, 2021-12-09:
I have read your book chapter on CrazyS. In this book chapter you do note the motor_constant, moment_constant, rotor_drag_coefficient and rolling_moment_coefficient for the Crazyflie, but you do not seem to say how you found these values.

Were they found experimentally? And if so, what equations were used for them?
Also, I notice that the rotor_drag_coefficient and rolling_moment_coefficient are the same as the default values from other RotorS model sdf's. Are these values not important for a good simulation result?

There has been extensive discussion on the underlying math of these parameters in another repo based on RotorS (https://github.com/PX4/PX4-SITL_gazebo/issues/110), but that discussion never quite reached a clear conclusion. I was hoping you might have the answer to these questions.

gsilano, 2021-12-17:
Hi @michieltebraake! As you stated on page 6 (this is the link to the preprint file, https://giuseppesilano.net/publications/rosChapter19.pdf), the parameter values of the Crazyflie have been taken from the repository [46] by the same research group of [29]. These parameters have been never validated with a real-world experiment. Hence, for the sake of completeness, Forster's parameters have been also implemented into the repository (see #56).

[46] The Automatic Coordination of Teams Lab, “GitHub Repository, RotorS fork, crazyflie-dev branch.” [Online]. Available: https://goo.gl/tBbS9G
[29] W. Honig and N. Ayanian, “Flying Multiple UAVs using ROS,” in ¨ Robot Operating System (ROS): The Complete Reference (Volume 2), A. Koubaa, Ed. Springer International Publishing, 2017, pp. 83–118.

gsilano, 2022-04-16:
Closed for inactivity.

closed 2022-04-16 by gsilano