gsilano/CrazyS #85: Quadcopter not flying

Flautdn, 2021-11-02:
Hello!

I have installed gazebo 9+ Ros melodic on ubuntu 18.04. You have helped me previously with the catkin build process, issue #84.

 roslaunch rotors_gazebo crazyflie2_hovering_example.launch is successfully launching the quad copter in the world but its not hovering. I have gone through other issues where you have made changes in the script and updated it on master. 

I doubt if  hovering example node is  publishing the trajectory to follow. Please help to fly this drone.

gsilano, 2021-11-04:
Hi! Try with #69, #63, #50, and similar. I think that's an issue related to some missing links.

Flautdn, 2021-11-05:
Hi @gsilano

Thank you so much for your help! The issue was with some of the incomplete gazebo packages

$ sudo apt-get remove ros-melodic-gazebo* gazebo*
$ sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
$ wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
$ sudo apt-get update
$ sudo apt-get install gazebo9 gazebo9-* ros-melodic-gazebo-*
$ sudo apt upgrade

As mentioned on readme, these commands fix the problem. Thank You :D

closed 2021-11-05 by Flautdn