gsilano/CrazyS #87: Leap Motion Integration

dingthe4th, 2021-11-22:
I just started learning about this project (thank you for this). As I have successfully tested the virtual keyboard joystick, I am wondering if is it possible to integrate `[Leap Motion Controller](https://www.ultraleap.com/product/leap-motion-controller/)` (hand gesture controller) for CrazyS. If so, what are the initial steps needed to accomplish this?

gsilano, 2021-11-27:
Bye! Well, this should be tested, but if a ROS package is distributed by the company or is available on the web, and you design your own wrapper to encode messages from the sensor to actions (e.g. pitch, push, roll and yaw rate), I don't see why it shouldn't work. Stay tuned because I new release of CrazyS for ROS Neotic on Ubuntu 20.04 is coming soon.

dingthe4th, 2021-11-28:
Thank you for this, I will study about it Sir. Looking forward for the release for Ubuntu 20.04!

closed 2021-11-28 by dingthe4th

gsilano, 2021-12-28:
CrazyS is now available for ROS Noetic and Gazebo 11 on Ubuntu 20.04 (#89)