gsilano/CrazyS #88: using the packages with ros noetic

danielemorra98, 2021-11-27:
Hi, 
I mounted Ubuntu 20 on my machine, so I'm trying to build and install the packages needed for the simulation in a ROS noetic distributions, but I run into some issues: I didn't manage to install the various `ros-noetic-rotors-*` packages, it seems that they do not compare in the system package manager (even if I found [them](https://aur.archlinux.org/packages/ros-noetic-rotors-evaluation/) on an archLinux distribution. So skipping the installing commands in the README.md, when I build the project with catkin I obtain this dependency error.
[image]
Checking the [Cmakefile](https://github.com/gsilano/CrazyS/blob/master/rotors_gazebo_plugins/CMakeLists.txt#L117) it couldn't be a missing dep so I figured out it should be a problem of skipping the install steps. Doo you have any ideas or suggestion?

gsilano, 2021-11-27:
Hi! I'm aware of the limitation of CrazyS on Ubuntu 20.04 and ROS Neotic. I just created a branch for this (`dev/ros-neotic`). During this Christmas holiday, I will be working on it. I'll send a message and update the repository as soon as it's ready. Keep in touch and stay tuned ;-)

danielemorra98, 2021-11-28:
Thank you, I'm wondering if I may help you, since I need a simulation environment of a crazyflie for a university project and I would like to improve my coding skills. Let me know if I can do something in the meantime

gsilano, 2021-11-28:
Well, you could, but this will take you a while. Perhaps, a couple of weeks. if you wanna try, you should install RotroS as described in [this pull request](https://github.com/ethz-asl/rotors_simulator/pull/651) and then integrate the current version of CrazyS for ROS Melodic (i.e., `dev/ros-melodic` branch). If successful, I would appreciate if you could open a pull request for this. Enjoy and good luck!!

danielemorra98, 2021-11-29:
I have "exported" (brutally copied and pasted into [rotor_simulator](https://github.com/ethz-asl/rotors_simulator)) the core parts needed to the `crazyflie2_hovering_example.launch`, I have built it (together with the [mav_comm](https://github.com/gsilano/mav_comm/tree/med18_gazebo9) repo) and it works. Now I think I should recognize the changes at that repo to create a pull request. The most challenging part will be find an alternative to the 'apt install ros-noetic-*'

gsilano, 2021-12-28:
As stated in #89, I've been working on it these days. Now, CrazyS works on Ubuntu 20.04 and ROS Noetic. A new release will follow soon.

closed 2021-12-28 by gsilano