gsilano/CrazyS #90: Trouble with crazyflie2 hovering example. [Ubuntu 18.04, ROS Melodic, Gazebo 9]

umidah, 2022-02-01:
Good day, I installed CrazyS for ubuntu 18.04, ROS melodic and Gazebo9. When running the crazyflie2 hovering example, I could not get it to run.

I used the example command given in the readme:
```bash
roslaunch rotors_gazebo crazyflie2_hovering_example.launch
```

This was the error displayed for me.
```bash
... logging to /home/ajtpa/.ros/log/cdbdd2ee-835f-11ec-989a-e0d55ea1eb65/roslaunch-otonokizaka-11207.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
name 'namespace' is not defined 
when evaluating expression 'namespace'
when processing file: /home/ajtpa/catkin_ws/src/CrazyS/rotors_description/urdf/component_snippets.xacro
included from: /home/ajtpa/catkin_ws/src/CrazyS/rotors_description/urdf/crazyflie2_base.xacro
included from: /home/ajtpa/catkin_ws/src/CrazyS/rotors_description/urdf/mav_generic_odometry_sensor.gazebo
RLException: while processing /home/ajtpa/catkin_ws/src/CrazyS/rotors_gazebo/launch/spawn_mav_crazyflie.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [['/opt/ros/melodic/lib/xacro/xacro', '--inorder', '/home/ajtpa/catkin_ws/src/CrazyS/rotors_description/urdf/mav_generic_odometry_sensor.gazebo', 'enable_logging:=false', 
```

If any more information is needed, I would gladly provide it.

rglbiv, 2022-02-01:
Hello, I got the same issue that returns the same error message. I followed the instruction in the README.md for Ubuntu 18.04.

gsilano, 2022-02-01:
It seems that the problem lies in some hold definitions contained in the `spawn_mav_crazyflie.launch`. Open the file, and add the `--inorder` option as described in here https://answers.ros.org/question/354092/xacro-in-order-processing-for-husky_gazebo/. I hope this helps.

umidah, 2022-02-01:
Thanks for the reply, I'll check it out. If it does not work, I might try re-installing with ubuntu 20.04 and ROS noetic.

rglbiv, 2022-02-01:
Thanks for the quick reply. Unfortunately, the --inorder argument is already in the file as seen below.
[image]

If I try to launch the file with or without the --inorder, the same error appears.
[image]

gsilano, 2022-02-01:
Looking at your screens, especially the second, it seems there is a problem with the namespace. Could you try to run any other launch file in the repository? I would like to understand if this problem affects only the file you mentioned or all the scripts have suffer of the same issue. Thanks!

rglbiv, 2022-02-01:
I tried it with:
mav_with_keyboard.launch and crazyflie2_swarm_hovering_example.launch, which also suffers from the issue.
[image]

gsilano, 2022-02-01:
Ok, so everything seems to be right with the ROS environment. Have you tried to run the Crazyflie example without the controller?

rglbiv, 2022-02-02:
If I try to run the file and comment the controller part, this is the result:
[image]

gsilano, 2022-02-02:
Ok! Let me test it on my PC. I wil give you back in a short time.

rglbiv, 2022-02-05:
Good day sir, are there updates to a fix to this issue? Thanks!

rglbiv, 2022-02-08:
I tried reinstalling it again in a fresh machine, and the error is still there. Problem regarding namespace.
[image]

rglbiv, 2022-02-08:
I tried installing it with Ubuntu 20.04 and it works with no issues, I guess there's a problem within Ubuntu 18.04. Thanks for your time.

gsilano, 2022-02-08:
Ok great! Did you compare the launch files? I'm curious if the problem lies there, or is it just a problem with ROS. However, this incompatibility problem is new to me.

rglbiv, 2022-02-08:
Unfortunately no as I have already deleted my Ubuntu 18.04 to install the new version.

gsilano, 2022-02-08:
Ah, okay. Finally you managed to make it works, and this what really matters ;-) Enjoy!

closed 2022-02-08 by gsilano

anavsalkar, 2022-02-12:
Hello, I am facing the same issue on my 20.04 system. I correctly installed as per the instructions for Ubuntu 20.04.
[image]
I tried changing the "<param name="robot_description"......" in the "spawn_mav.launch" file including/ommiting '--inorder' or 'xacro' or 'xacro.py' as suggested in ( https://answers.ros.org/question/122021/xacro-problem-invalid-param-tag-cannot-load-command-parameter-robot_description/ ), without any success. 
What's surprising is that I had no trouble working with RotorS installation from 'eth-asl' when I last used it 2 months back. Now, both the installations (earlier RotorS from eth-asl and CrazyS) give the same errors. Don't know if some update has caused this.
Thank you in advance for any help!

anavsalkar, 2022-02-13:
I managed to solve the issue. Well, like I said it was indeed due to version updates. I removed `xacro` and `xacro-1.14.11.egg-info` directories from `/opt/ros/noetic/lib/python3/dist-packages` folder using `sudo rm -r ...` command and installed the 1.13.3 version of xacro using `pip3 install xacro==1.13.3`. This simple downgrading from 1.14.11 to 1.13.3 worked for me, without any changes in launch files.

gsilano, 2022-02-13:
Great! Thanks for letting the community know about this. I'll add these instructions to the README.md file in case someone else experiences the problem.

dogukanteber, 2022-02-13:
Thanks a lot @anavsalkar. You saved me from a headache. Your solution worked.